Atmosphere-NX/Atmosphere #2420: Override button/API to launch application/program without update NCAs

fruityloops1, 2024-10-26:
## Feature Request

### What feature are you suggesting?
#### Overview:
When starting a program, this feature would add a way to avoid the update NCAs being layered atop the base NCA, so the program is launched in it's "base" version instead.

#### Smaller Details: 
- This could be made exclusive to Applications with a button override, or
- be made available through some api (custom service? or extension of an existing one) that allows you to set a flag for a specific Program ID to be launched without updates as long as that flag is set

#### Nature of Request:  
- Addition

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br> / ncm?/fs?/ldr? not sure

### Why would this feature be useful?
For programs bundled with HOS, this wouldn't be very useful, but for Applications it would be. For example: Super Mario Odyssey modders tend to prefer version 1.0.0 of the game as the latest version does not export symbols, so they resort to downgrading their game using dumps of the 1.0 game piped through LayeredFS with an exefs.nsp and romfs.bin, or straight up deleting the update using some piracy tool, to then later redownload the update from the eShop when needed. With this feature, there would be way less hassle in launching version 1.0.0 of the game, and no need to mess around with title installs or huge layeredfs dumps on the SD card.

DaDev123, 2024-10-26:
W

Amethyst-szs, 2024-10-26:
That example of the use case for Super Mario Odyssey would be incredibly useful, especially for inexperienced players/modders that struggle with mod setup, keys, and the Odyssey Downgrader.

Would it be possible this could be integrated into the atmosphere/contents/<titleid>/ folder? The ability to include some form of "version config" with romfs/exefs mod downloads to ensure the end user is running the right version of the software would be incredibly useful.

Either way, really hope to see this feature someday!

LynxDev2, 2024-10-26:
I think most people in the smo modding community agree that adding this feature would be really nice

SciresM, 2024-10-27:
Ignoring the brigading, I am not, I guess, fundamentally opposed to this, but I'm also not actually sure what would be involved in implementing it. 

I guess this would probably be loader changes...have you done any research on what specifically would be needed? This is a lot more likely to happen if presented to me as a set of instructions or e.g. a git diff or a pull request.

fruityloops1, 2024-11-08:
As far as ive researched this would require patching FS (loader could stay untouched) but patching is probably too hacky for the amount of code that would need to be changed so it would have to be reimplemented and I don't wish to pick that up as a task myself since I think it would be a decent amount of work (and I have no clue how kips work idk how to even get the FS binary), so I don't think it will be happening (unless somebody does want to reimplement FS)

closed 2024-11-08 by fruityloops1