Atmosphere-NX/Atmosphere #2421: 2168-0002 (0x4a8)

SignorUpB, 2024-10-26:
### What's the issue you encountered?

I accidentally updated my switch's firmware without updating the atmosphere version (sysmmc no hekate) and after getting out my sd and replacing it with the new pre-release (and replacing the fusee.bin file) it keeps giving me this error whenever I try to turn on my switch with atmosphere.

### How can the issue be reproduced?

Update switch firmware with latest stable release;
Replace CFW with the 1.8.0 on the micro sd card;
Send fusee.bin payload with TegraRcmGui 2.6 

### Crash Report

(https://gist.github.com/SignorUpB/fe2e6708c08fc03826cec867d0af10b1)

### System Firmware Version

19.0.0

### Environment?

- Launched atmosphere with fusee.bin payload with TegraRcmGui
- 1.8.0 Pre-release atmosphere
- Also running:
    -Sys-Clk
    -Emuiibo
    -Fizeau
    -Custom These with NXThemesInstaller
- Tesla Menu Installed 
- SysNAND:

masagrator, 2024-10-26:
Remove theme.

tl;dr remove "0100000000001000" folder from atmosphere/contents

closed 2024-10-27 by SciresM

marley-crazy, 2025-12-18:
> Remove theme.
> 
> tl;dr remove "0100000000001000" folder from atmosphere/contents

not working