Atmosphere-NX/Atmosphere #2423: Switch keeps crashing after a few seconds in the home screen when wifi is on

kolibiri, 2024-10-27:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

borntohonk, 2024-10-27:
https://github.com/Atmosphere-NX/Atmosphere/issues/2215

disable "blank_prodinfo_" in exosphere.ini

or hekate_ipl1.ini as "cal0blank"

or undo "incognito"

Rikj000, 2024-10-27:
> #2215
> or undo "incognito"

I've been running with an incognito serial (`XAWVAMPITECH00`) without issues on FW 17.0.0 + AMS 1.6.2,   
and nowadays on FW 19.0.0 + AMS 1.8.0-pre-c6014b533.   

However if you do not block Nintendo's servers through altering your DNS,   
then enabling WiFi connection will crash your system.

I have a hunch that they black-listed commonly used incognito serial numbers,   
to make the device crash when it's detected.

To keep incognito and prevent crashes,   
you can either alter your DNS to 90DNS (Blocks Nintendo servers):
- With https://github.com/suchmememanyskill/switch-90dns-setter/releases
- Or manually, configure to:
    - Primary DNS: 207.246.121.077
    - Secondary DNS: 163.172.141.219

Then you can validate afterwards with: https://github.com/meganukebmp/Switch_90DNS_tester/releases

I'm using a now deleted fork of Incognito RCM by @mistervampi,   
It's his `v0.7.4`, released on `2023-02-23`,   
I can share the payload `.bin` if allowed and if anyone interested.

bitelaserkhalif, 2024-10-28:
Try using [DNS-mitm](https://gist.github.com/dimaskiddo/bb803c5e008002214f9da917cfd1485f) feature and block any Nintendo servers. It's stronger and much more resilient alternative than 90dns, since it still in effect even after Wi-Fi switching.

zachyyc, 2024-11-06:
ive tried all these things and im still getting the error screen, could it be anything else?

sthetix, 2024-11-09:
try using my pack and see, https://www.sthetix.info/hats/releases/

zachyyc, 2024-11-09:
Im using your pack currently. I ended up doing the reset (unbricking vol2)
then I downgraded to the fw before. 19.0.0. everything was smooth until I
was in hekate then I hit power down and I haven't been able to get it
passed a black screen since then

SciresM, 2024-11-09:
Piracy is not supported, do not advertise for it.

closed 2024-11-09 by SciresM