Atmosphere-NX/Atmosphere #2424: Software Manager Tinfoil Crashes when opening through Home Menu.

BenelloDonatello, 2024-10-27:
## Bug Report

### What's the issue you encountered?

When trying to open Tinfoil through the Switch's Home Menu the Software closes immediatly with the Error "The Sofware could not be started.". 

I did not make any changes to Atmosphere other than installing the newest sigpatches form this gbatemp download: 
https://gbatemp.net/attachments/hekate-ams-package3-sigpatches-1-8-0p2-cfw-19-0-0_v4-zip.465290/

### How can the issue be reproduced?

Opening Tinfoil from the Switch's Home Menu with or without internet connection.

### Crash Report

Atmosphere did not crash.

### System Firmware Version

19.0.0
and
18.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: **hekate**
- Official release or unofficial build: **official**
  - version: **6.2.2**
- Do you have additional kips or sysmodules you're loading: **yes**
  - sysmodules: **sysclk** (loaded but not active); **emuiibo** (loaded but not active)
  - kip1patch: **nosigchk**
- Homebrew software installed: hbmenu official v3.6.0; nx-hbloader official v2.4.4
- EmuMMC or SysNAND: **emuMMC**
  - **partition-based**

### Additional context?

- Additional info about your environment: **-none-**
- the Applet Version of Tinfoil still works

Aura67, 2024-10-27:
Piracy is not supported.

AntiGrav1ty, 2024-10-28:
@BenelloDonatello this is no Bug from Atmosphere.
This is due to the NX Forwarder which has to be created new for the v19. 
You can just replace the nro with the applet version, and start through elevated menue.

BenelloDonatello, 2024-10-28:
Got it. Thank you for your Time. I'll be sure to dive depper into an issue on the internet before creating a bug report next time.

TechnoidAlpha, 2024-10-29:
> @BenelloDonatello this is no Bug from Atmosphere. This is due to the NX Forwarder which has to be created new for the v19. You can just replace the nro with the applet version, and start through elevated menue until they release an updated selfinstaller version (nx forwarder) You could have also read this by reading some threads on the mentioned website that you posted 😁

I had to remake all my forwarders for the latest version of atmosphere. I used a fork of the "online NRO to NSP forwarder" that worked for tinfoil applet mode nro:

https://nsp-forwarder-git-fork-masagrator-main-tootallteam.vercel.app/

I haven't tried the forwarders on older firmware but this worked for my tinfoil and other HB apps.

Hopefully that helps!

cosmii02, 2024-11-10:
The no brain dislike emojis are wild, crying over people using tinfoil

impeeza, 2024-11-10:
Piracy is not supported please close the issue

cosmii02, 2024-11-10:
> Piracy is not supported please close the issue

What is this crying, aintnoway

closed 2024-11-10 by SciresM