Atmosphere-NX/Atmosphere #2425: stuck in a black screen while in sleep mode

Astr4l1ty, 2024-10-27:
### What's the issue you encountered?

When my screen automatically turns off, it stays off until I restart my Switch by holding the power button.

### How can the issue be reproduced?

Just put your Switch in sleep mode

### Crash Report

It didn't save any crash reports when that happened.

### System Firmware Version

19.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Official
  - version : 6.2.2
- Do you have additional kips or sysmodules you're loading: yes
    sysmodules: sysclk, sys-patch
- Homebrew software installed: hbmenu official v3.6.0; nx-hbloader official v2.4.4
- EmuMMC or SysNAND: EmuMMC
  - partition based

### Additional context?

This happened on a modchipped OLED
I tried disabling additional kips and sysmodules
I didn't tested on file-based, I will try
I have 19.0.0 installed as OFW and CFW

impeeza, 2024-10-27:
Seems is a problem of FW 19 not of Atmosphère:

https://gonintendo.com/contents/41253-switch-firmware-update-19-0-0-now-available-updates

Astr4l1ty, 2024-10-27:
Well your right bcoz it just happened to me without Atmosphère...
I hope Nintendo will fix this asap.

WestonGuidero, 2024-10-28:
I updated Sysclk to the latest and it fixed it for me.

schizofrog, 2024-10-28:
I deleted Atmosphere, Bootloader, Sigpatches and reinstalled them, I no longer black screen in sleep and it works fine for me

Astr4l1ty, 2024-10-28:
Fixed it by updating sys-clk thanks @WestonGuidero

closed 2024-10-28 by Astr4l1ty