Atmosphere-NX/Atmosphere #2426: NSP to NRO forwarders not working

resurrecta, 2024-10-27:
Since the change to the 1.8.0 pre-release, the forwarders no longer work.
https://nsp-forwarder.n8.io/
These used to work well on the previous version. This means right now that for example the Homebrew Store can't be displayed on the main menu anymore.
When I try it, just crashes... Black screen, it can not be started.

SciresM, 2024-10-27:
Piracy is not supported.

closed 2024-10-27 by SciresM

resurrecta, 2024-10-28:
Hey, I normally don't fall in discussions over piracy. But could you explain how building a forwarder with own keys to reference the homebrew app itself is considered piracy?

Thank you for your attention

Closed #2426<https://github.com/Atmosphere-NX/Atmosphere/issues/2426> as completed.

Masamune3210, 2024-10-28:
It requires sigpatches, its considered piracy.

SciresM, 2024-10-28:
All installed content requiring signature patches is indistinguishable from piracy to Atmosphere (and  Nintendo), so support is provided for none of it.