Atmosphere-NX/Atmosphere #2427: Sleep Mode Not Functioning

Beyondzxc, 2024-10-29:
## Bug Report

### What's the issue you encountered?

[ I've updated to the latest atmosphere and for a while was encountering no issues. A few days later, everytime I go into cfw and go into sleep mode, it never wakes up and I have to hold the power button to reset it.]</br>
[ No changes were made to atmosphere. ]</br>
[N/A]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Downloading the latest Atmosphere seems to be the only cause for this issue.]</br>

### Crash Report

[N/A]</br>
[ N/A ]</br>
[N/A]</br>

### System Firmware Version

19.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: 
  - [ pre-release build ]
  - [N/A] 
     - [ N/A ]
     - [ N/A ]
- Do you have additional kips or sysmodules you're loading:N/A
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ SysNAND]

### Additional context?

- Additional info about your environment:
- [ N/A ]

impeeza, 2024-10-29:
it's a HOS problem not atmosphère AS HAS BEEN STATED ON THE OTHERS ISSUES, please wait for Firmware 19.0.1 and the proper Atmosphère support to it.

ffffalcao, 2024-11-01:
updating sys-clk fixed it for me

Teottry08, 2024-11-01:
> updating sys-clk fixed it for me

Could you please help me with that?

impeeza, 2024-11-01:
This is not Atmosphère issue. Please close the issue and go to GBATemp or Discord for support

KingKarlos2020, 2024-11-01:
I'm on 19.0.0 and it boots to OFW in seconds and the sleep mode works. When I use AMS1.8.0 its takes along time to boot and if it goes to sleep mode it stays black screen and sometimes just reboots to hekate.

ffffalcao, 2024-11-01:
> > updating sys-clk fixed it for me
> 
> Could you please help me with that?

Assuming that you already have sys-clk installed on your switch and that the cause of the problem is the same mine was

Go to sys-clk github and download the latest release (sys-clk-2.0.1.zip), extract it and follow the instructions on the Readme file

Paradox625, 2024-11-21:
> updating sys-clk fixed it for me

This fixed it for me. Just go to sys-click release page download latest and drag and drop files if you already have sys-click setup from the past firmware

closed 2025-04-03 by SciresM