Atmosphere-NX/Atmosphere #2428: Error 2182-0000

Ganesh2556, 2024-10-29:
Hello. Error 2182-0000 (0xb6) appeared with process ID 0100000000000045. Firmware 18.1.0 (Atmosphere 1/7/1 -master-539491f5d). I am attaching a crash report to the message. I did not make any changes to the firmware. Just playing games.

[01730171493_0100000000000045.log](https://github.com/user-attachments/files/17551160/01730171493_0100000000000045.log)

[01730171492_0100000000000045.log](https://github.com/user-attachments/files/17551161/01730171492_0100000000000045.log)

masagrator, 2024-10-29:
1. You didn't use issue template
2. Looks like a hardware issue, not an Atmosphere bug.

Ganesh2556, 2024-10-29:
Sorry for not using the issue template. I had a hard time filling it out. 
Everything is fine with the hardware, since the stock firmware starts without problems. The error occurs in the firmware launched from SD, and it doesn’t matter - safe mode or just launching the atmosphere.

neomhor, 2024-10-30:
Maybe a corrupt SD or EmuNAND

closed 2025-04-03 by SciresM