Atmosphere-NX/Atmosphere #2429: Fatal Error (Program ID 690000000000000d

illuminatusdeus, 2024-10-29:
Faced issue of fatal error / crash. Atmosphere does not boot at all.

[image]
[report_000000001ca36d3b.zip](https://github.com/user-attachments/files/17557685/report_000000001ca36d3b.zip)

I deleted the folder from Atmosphere content folder and the problem was resolved. Just thought I'd report this problem to devs.

borntohonk, 2024-10-29:
> I deleted the folder from Atmosphere content folder and the problem was resolved. Just thought I'd report this problem to devs.

You can go ahead and close this issue as your report is filed in the wrong github repository. The error you're experiencing belongs to sys-con, which i belive current maintainer is this one https://github.com/o0Zz/sys-con/issues

Ishi25, 2024-11-15:
Thank you, it deletting the folder solved the problem for me too

illuminatusdeus, 2024-11-16:
I am closing the issue, as the problem has been solved and as advised. For the same problem, do use this thread as reference to resolve this problem if faced in the future. Thank you to all contributors in the scene!

closed 2024-11-16 by illuminatusdeus