Atmosphere-NX/Atmosphere #2431: 新的报错2123-0011

tuanzi11, 2024-11-02:
![上传4264c1d06cb484f4112dacacbc952921.jpeg...]（）各别游戏提示2123-0011报错代码      
大气层1.8.0 系统19.0.1

tuanzi11, 2024-11-02:
[image]

fmkli, 2024-11-02:
首先，英文提出issues
第二，请提供更多信息
第三，关于 2123-0011 的错误信息网上内容有很多，如果可以，请先了解一二

tuanzi11, 2024-11-02:
I also reinstalled the game, changed the integration package, entered maintenance mode and turned off flight mode! Some games cannot progress! Tip 2123-0011

tuanzi11, 2024-11-02:
I have tried many ways, but no solution!

Tetsuya-Takashi, 2024-11-02:
The error code 2123-0011 might actually be this and not related to Atmosphère itself: https://en-americas-support.nintendo.com/app/answers/detail/a_id/48626/~/error-code%3A-2123-0011

impeeza, 2024-11-02:
Piracy is not supported, you are trying to load a software for which the console have no the needed certificates nor tickets, please for support go to Discord or GBATemp.

please close this issue is not Atmosphère related.

closed 2025-04-03 by SciresM