Atmosphere-NX/Atmosphere #2432: Bug Report: Error Code 2123-0011 When Launching Games on Atmosphere CFW

Spenzeus, 2024-11-02:
![WhatsApp Image 2024-11-02 at 20 58 43_f52e3ed5](https://github.com/user-attachments/assets/9f564932-8a96-4425-874f-Bug Report
What's the issue you encountered?
I’m running into error code 2123-0011 on my Nintendo Switch with Atmosphere CFW. This error appears every time I try to enter games, and it tells me to restart the console from the power options. Before this started happening, I deleted the Nintendo server blocks, and even after restoring them, the problem didn’t go away.

How can the issue be reproduced?
Boot the Nintendo Switch into Atmosphere.
Try to launch any game.
Receive the error message and need to restart the console.
Crash Report
I didn’t find a specific crash report in /atmosphere/crash_reports, but the console crashes consistently when attempting to enter games, leading to the error code.

System Firmware Version
Firmware Version: 19.0.1

Environment?
Bootloader: Hekate
Atmosphere Version: atmosphere-1.8.0-prerelease
Additional kips or sysmodules: None that I know of
Homebrew software installed: Awoo Installer, Goldleaf
EmuMMC or SysNAND: Using try in both
Additional context?
I formatted my SD card and reinstalled all the Atmosphere and Hekate files to ensure everything is fresh. I also tried using the Incognito tool to hide my console’s serial number, but I’m still facing this error when trying to launch games. I’ve double-checked my network connection and DNS settings—everything looks fine, and I even switched to manual DNS to block Nintendo’s servers.

I’d really appreciate any help to resolve this issue!

Masamune3210, 2024-11-02:
Piracy is not supported.

Spenzeus, 2024-11-02:
what do you mean? this is all the point of cfw

Tetsuya-Takashi, 2024-11-02:
The error code 2123-0011 is this and is not related to Atmosphère itself: https://en-americas-support.nintendo.com/app/answers/detail/a_id/48626/~/error-code%3A-2123-0011 people need to learn how to google error codes first before you assume it's a bug with Atmosphère.

Masamune3210, 2024-11-02:
> what do you mean? this is all the point of cfw

An excerpt from the template that you obviously didn't read 
[image]

impeeza, 2024-11-02:
> what do you mean? this is all the point of cfw

Are you insane, CFW allows you to run Homebrew apps, to use your console with the programs made by the community.  if you want piracy go else where, please close the issue.

curcius, 2024-11-09:
> > what do you mean? this is all the point of cfw > > Are you insane, CFW allows you to run Homebrew apps, to use your console with the programs made by the community. if you want piracy go else where, please close the issue.

 These types of comments lead me to believe that Nintendo is winning over Atmosphere, 19.x.x came out and until now it hasn't left the pre-release so I think it's the beginning of the end of Atmosphere.

Masamune3210, 2024-11-09:
.....just because it hasn't left pre-release doesn't mean that things are coming to an end. Turns out, people have these things called lives

impeeza, 2024-11-09:
> These types of comments lead me to believe that Nintendo is winning over Atmosphere, 19.x.x came out and until now it hasn't left the pre-release so I think it's the beginning of the end of [...]

Seriously...  please, please, please read the main page AND THE RELEASE page SciresM release the most recent version as «prerelease» because Atmosphère is always a Work In progress, by the way made on the spare time of the developers, you should be d; anyway, is a prerelease because is needed the community test it and feedback improvements AND REAL ISSUES. sometimes the prerelease tag last until next release.  If you see the issue for work needed for FW 19, you will see some things still are on development.  so yes, PRERELEASE.

Tetsuya-Takashi, 2024-11-12:
The error code 2123-0011 has nothing to do with Atmosphère, do people not know how to google error codes these days?

WrittenSkate, 2024-11-14:
you should've waited and kept at 18.0.0, I still haven't updated to this day

Angry0tter, 2025-03-16:
I was having the same issue and it looks like the cause is save data cloud enabled on the Nintendo account linked to emummc.  Once I used goldleaf to unlink the account the error stopped occurring.

closed 2025-04-03 by SciresM