Atmosphere-NX/Atmosphere #2433: Fusee not working with physical RCMloader

Apache811, 2024-11-03:
## Bug Report

<br>

### What's the issue you encountered?

As per title, the latest fusee 1.8.0-prerelease, doesn't boot the switch in atmosphere if used on a RCMloader: the switch is stuck on a black screen and I need to hold the power button to force a shut down.</br>
Using the 1.7.1 causes an error about a package not found but then the reboot starts the console in Atmosphere.<br>
I didn't make any changes to Atmosphere.</br>

### How can the issue be reproduced?

</br>
Have the switch shut down, plug the RCMloader in the usb-c port, slide the jig in the right joy-con slot, hold the volume up button and press the power button. The console will be stuck in a black screen after the RCMloader led will turn on and off failing the ijection for some reasons.<br>
As mentioned, using the previous version of fusee without changing anythis result in a succesful injection BUT NOT in a succesful boot.</br>

### Crash Report

N/A</br>

### System Firmware Version

19.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee with latest heckate, all official versions.

### Additional context?

- Additional info about your environment:
- N/A

borntohonk, 2024-11-03:
This isn't really an atmosphere issue, you have to swap out the fusee.bin...

or you can use hekate to chainload atmosphere's fusee bin with an entry such as this, as the atmosphere zip always contains the latest fusee.bin as reboot_payload.bin (much more convenient for your usecase)

```ini
[cfw]
payload=atmosphere/reboot_payload.bin
```

Apache811, 2024-11-03:
I am really sorry, but I don't think I understood your answer.
I swapped the fusee.bin on my RCMloader with the lates 1.8.0-prerelease version BUT, as I said, when it is time to inject the payload, the RCMloader seems to not work correctly: everything it's stuck on a black screen.
This is the first time that happened with me, therefore, I thought it was a fusee problem: maybe a compatibility issue with the RCMloader because if I use the TegraRCMGUI everything works correctly.
again, as mentioned in my comment, if I use the fusee version 1.7.1, it will fail because the Atmosphere version installed on the SD is the 1.8.0-prerelease but it will boot in Atmosphere anyway probably because it automatically uses as fallback the reboot_payload.bin you were talking about.

I still believe there is something in the latest fusee.bin that it is not compatible with a usb RCMloader.

If this doesn't make any sense and I am wrong, I apologize.

borntohonk, 2024-11-03:
> I still believe there is something in the latest fusee.bin that it is not compatible with a usb RCMloader.

It is not an atmosphere issue, the solution mentioned above would resolve your problem anyway.

Goldenwolfz21, 2024-11-04:
hi ive had this problem
make sure to rename the latest fussee.bin to payload.bin
and have the right color on the rcmloader to inject it

Akira808State, 2024-11-05:
Regarding the RCM loader that plugs into the USB-C port, press and hold the small "+" button. If it flashes a different color, then it's in the incorrect folder. Press and hold that button until it flashes blue for Atmosphere. It should then inject the payload.

Apache811, 2024-11-05:
Hello everyone and thank you very much for all your comments and suggestions.
I have been using the RCMloader for a while now and I have updated the fusee already several times... it's just that, for me at least, the latest version of the fusee doesn't work and I wanted to report this issue.   :)

impeeza, 2024-11-05:
For your RCM Lodaer use ArgonNX-SE payload AND FORGET to upgrade your RCM Loader never again, only update files on your SD Card.

Yahya91377, 2024-11-16:
Hey ok I'm booted into hekate when but when trying to boot emcmm..it states on the screen emummc is forced but not enabled also failed to launch hos

[image]

impeeza, 2024-11-16:
its a EmuNAND configuration issue, get inside NYX (the Hekate's GUI) and setup your EmuNAND.

please close this issue.

closed 2024-11-16 by SciresM

sennastien, 2025-01-15:
I do have exactly the same issue as mentionned above,

Everything worked so far correctly (it's been years I've been experiencing explode and so on with the payload.bin to update)

Apart from using chain load as said above, there's a compatibility issue with the latest fusee.bin and the rcmloader for this release.