Atmosphere-NX/Atmosphere #2434: Bug Report: 2162-0002(0x4a2)

1149271370, 2024-11-04:
## Bug Report
2162-0002(0x4a2)(which is different from usual) 

### What's the issue you encountered?

After disabling Airplane Mode or attempting to connect to Wi-Fi,The system crashes and prompts me to restart, regardless of whether it's the official system or an emulated one

### How can the issue be reproduced?

The issue can be reproduced simply by entering either the official system or the emulated system, then disabling Airplane Mode or attempting to connect to Wi-Fi.

### Crash Report

[01730689576_010000000000000e.log](https://github.com/user-attachments/files/17613463/01730689576_010000000000000e.log)
[01730689576_010000000000002f.log](https://github.com/user-attachments/files/17613466/01730689576_010000000000002f.log)

### System Firmware Version

19.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate 6.2.2
- Official release or unofficial build: 
  - [  unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation by other]
- Do you have additional kips or sysmodules you're loading: I'm not sure how to verify this option
- Homebrew software installed: [ DBI,DayBreak ]
- EmuMMC or SysNAND:EmuMMC include file-based(maybe?)

### Additional context?

I apologize; this is my first time filling out a troubleshooting form, so my description may not be very accurate. However, I’m eager to resolve this issue and am trying to figure out where the problem lies.

impeeza, 2024-11-04:
Enable dns MiTM or remove prodinfo blanking

1149271370, 2024-11-04:
> Enable dns MiTM or remove prodinfo blanking

[image]
dns MITM seems like it might already be enabled
I will try deleting the files now.

1149271370, 2024-11-04:
> ![image](https://private-user-images.githubusercontent.com/83260157/382637684-2e240e3e-d6df-4373-91b1-2f27148a18a0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

[image]

Is it this file in this directory? I deleted it, but the same error still occurs.But I still really appreciate you answering my questions.

HeikesFootSlave, 2024-11-04:
> ![image](https://private-user-images.githubusercontent.com/83260157/382638640-6d6b1c56-f705-4502-acfc-1939e43d12bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

You deleted the file? Standing Ovations! WHY?
Hope it creates it again at startup because it is in a BACKUP Folder!!! BACKUP!

@impeeza NEVER said delete those files he said enable dns_mitm in system_settings or remove prodinfo_blanking = exosphere.ini in the root of the SD (if not get it from ams folder in config_templates)

1149271370, 2024-11-04:
> You deleted the file? Standing Ovations! WHY? Hope it creates it again at startup because it is in a BACKUP Folder!!! BACKUP!
>
> @impeeza NEVER said delete those files he said enable dns_mitm in [...]

I'm very sorry; I misinterpreted what @impeeza meant 
Don’t worry, I have a backup of this backup file
Actually, while asking these questions, I handed my Switch over to a technician who can handle this kind of issue. I also shared the above suggestions with the technician, hoping they might help with the repair.
Thanks again to both of you. @impeeza @HeikesFootSlave

EGFNF, 2024-11-04:
> You deleted the file? Standing Ovations! WHY? Hope it creates it again at startup because it is in a BACKUP Folder!!! BACKUP!
>
> @impeeza NEVER said delete those files he said enable dns_mitm in [...]

A standing OVATION EVERYONE FOR THE BIGGEST JERK TECH SUPPORT GUY!

If you can't stand the heat STAY OUT OF THE KITCHEN my dude! Stop yelling at people and quit your day job if you can't handle lower intelligence people. Not everyone can be a "genius" like you!

HeikesFootSlave, 2024-11-04:
> I'm very sorry; I misinterpreted what @impeeza meant Don’t worry, I have a backup of this backup file Actually, while asking these questions, I handed my Switch over to a technician who can handle [...]

I have to apologies that was a bit harsh!
PRODINFO Data is very sensible and if you loose that and break it you are glad to have a backup!

1149271370, 2024-11-06:
Status update: The technician tried replacing some hardware, but it was to no avail, ultimately confirming that the issue lies at the software level. I also tried using some "all-in-one" packages supposedly supporting 19.0.1, but they didn’t work either. I’ve already seen similar cases appearing in some online groups, and this issue seems to be spreading. The technician told me that initializing the official system can resolve the issue, but since it’s impossible to connect to the internet, account data in the official system cannot be transferred to another device. Restoring data from the official (emulated) system to the official system carries a risk of getting banned. For now, I can only hope that Atmosphere will release an update soon and hope it might resolve this issue.

SciresM, 2024-11-06:
I don't know what's happening in this issue, but it's not an atmosphere problem. I would just fully wipe your nand following an unbricking guide.

closed 2024-11-06 by SciresM

1149271370, 2024-11-07:
The issue has been resolved. The solution was: after finding firmware compatible with 19.0.1, I updated the emulated system first, then updated the official custom firmware system, which allowed me to connect to the internet. The problem may have been caused by my own oversight, as my official custom firmware system had been previously blocked. Thank you all for your responses and sorry for the trouble—thanks again!

idkcode22, 2025-02-11:
If anyone else is having this problem but wants to use an old firmware this will fix it: https://github.com/Atmosphere-NX/Atmosphere/issues/2463#issuecomment-2509150084