Atmosphere-NX/Atmosphere #2435: Crash Report error code 2002-1004

conscript86, 2024-11-05:
## Bug Report

### What's the issue you encountered?

Getting error code 2002-1004 when DRAGON BALL Z: KAKAROT is executed. It only happens with this game.
The Nintendo screen shows the this message: The software closed because an error occurred.

### How can the issue be reproduced?

Same result after each time the game starts

### Crash Report

Atmosphère Crash Report (v1.7):
Result:                          0xD407D802 (2002-1004)

Process Info:
    Process Name:                AT
    Program ID:                  010051c0134f8000
    Process ID:                  0000000000000175
    Process Flags:               00000877
    User Exception Address:      00000036418eb000
Exception Info:
    Type:                        User Break
    Address:                     0000002a3ee624d4 (nnSdk + 0xfc4d4)
    Break Reason:                0x0
    Break Address:               00000070e522bbd8
    Break Size:                  0x4
Crashed Thread Info:
    Thread ID:                   000000000000ad4d
    Thread Name:                 GameThread
    Stack Region:                00000070e512c000-00000070e522c000
    Registers:
        X[00]:                   0000000000000000
        X[01]:                   00000070e522bbd8
        X[02]:                   0000000000000004
        X[03]:                   0000000000000000
        X[04]:                   0000000000000000
        X[05]:                   0000000000000000
        X[06]:                   0000000000000000
        X[07]:                   0000000000000000
        X[08]:                   0000000000000000
        X[09]:                   0000000000000000
        X[10]:                   0000000000000000
        X[11]:                   0000000000000000
        X[12]:                   0000000000000000
        X[13]:                   0000000000000000
        X[14]:                   0000000000000000
        X[15]:                   0000000000000000
        X[16]:                   0000000000000000
        X[17]:                   0000000000000000
        X[18]:                   0000000000000000
        X[19]:                   0000000000000000
        X[20]:                   0000000000000000
        X[21]:                   0000000000000000
        X[22]:                   0000000000000000
        X[23]:                   0000000000000000
        X[24]:                   0000000000000000
        X[25]:                   0000000000000000
        X[26]:                   0000000000000000
        X[27]:                   0000000000000000
        X[28]:                   0000000000000000
        FP:                      00000070e522ba20
        LR:                      0000002a3edbc518 (nnSdk + 0x56518) (_ZN2nn4diag6detail5AbortEPKNS_6ResultE + 0x18)
        SP:                      00000070e522ba20
        PC:                      0000002a3ee624d0 (nnSdk + 0xfc4d0)
    Stack Trace:
        ReturnAddress[00]:       0000002a3edbae10 (nnSdk + 0x54e10)
        ReturnAddress[01]:       0000002a3edbaed8 (nnSdk + 0x54ed8)
        ReturnAddress[02]:       0000002a3edee168 (nnSdk + 0x88168)
[... 294 lines cut ...]
                                 0036418eb680 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0036418eb690 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0036418eb6a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0036418eb6b0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0036418eb6c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0036418eb6d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0036418eb6e0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                                 0036418eb6f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

### System Firmware Version

18.1.0</br>

### Environment?

Running off Atmosphere 1.8.0 via fusee.bin
Hekate version v6.2.2
Switch v1

### Additional context?

I first installed a XCI file with Awoo that failed.
I try later with a NSP version with DBI.
I try installing only v0, with last update 1.51, with and without DLCs (always uninstalling the app before).
I also remove save files.

SciresM, 2024-11-05:
Piracy is not supported. The error is that it's pirated.

closed 2024-11-05 by SciresM

conscript86, 2024-11-05:
I also bought the game digitally in the eshop and have the same problem.