Atmosphere-NX/Atmosphere #2437: 0x4A8 (2168-0002)

ferrisargylezero, 2024-11-05:
## Bug Report

### What's the issue you encountered?

Upon installing a game I dumped with Tinfoil, the application crashed and gave me the error "The software was closed because an error occured"

I didn't make any changes to Atmosphere 

### How can the issue be reproduced?

Dump a game (specifically the legend of zelda tears of the kingdom) and then install it through Tinfoil. Tinfoil will crash, "The software was closed because an error occured" will be displayed and you will be sent to the home screen, and a crash log will be made in atmosphere/crash_reports

### Crash Report

https://gist.github.com/ferrisargylezero/f51564af3e63394f93a8e92ea121b57e

### System Firmware Version

19.0.1</br>

### Environment?

- Hekate 6.2.2
- Atmosphere Official release version 1.8.0
- Homebrew software installed: daybreak, ftpd, goldleaf, jksv, nx shell, nxthemes, tinfoil 18.0 applet version, nx-hbmenu,
- SysNAND

### Additional context?
Someone else has an issue open with the same error but it was not resolved.

borntohonk, 2024-11-05:
> Upon installing a game I dumped with Tinfoil, the application crashed and gave me the error "The software was closed because an error occured"

Piracy is not supported.

ferrisargylezero, 2024-11-05:
I dumped the game myself though? Trying to be able to play my own game without needing to have the cartridge in.

SciresM, 2024-11-05:
Piracy is not supported, use your cartridge, or buy the game on the eShop if you want to play it without the cartridge inserted.

closed 2024-11-05 by SciresM