Atmosphere-NX/Atmosphere #2438: Add an option to prevent the creation of logs on the microSD.

vencraft, 2024-11-06:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Disable the generation of error files from Atmosphere on the microSD.

#### Smaller Details: 
- Create an option in system_settings that allows disabling log generation on the microSD, thus avoiding continuous writing to the card and minimizing card degradation.

#### Nature of Request:  
- Addition
Add an option to prevent the creation of logs on the microSD.

### What component do you feel this would best fit within?
  - Atmosphère's custom bootloader.</br>

[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
Minimize the degradation of the microSD card by avoiding constant writes to the card.

Masamune3210, 2024-11-06:
The logs are there for a reason. Nothing should be continually making error logs because that would mean its continually encountering errors. If it is, find out why and fix it. I very much doubt that an option will be added

vencraft, 2024-11-06:
The biggest generation of errors I have is due to having Wi-Fi enabled and the Nintendo connections being blocked, which cannot connect in the background... Is there a way to avoid this?

hadrianrab, 2024-11-07:
> > The logs are there for a reason. Nothing should be continually making error logs because that would mean its continually encountering errors. If it is, find out why and fix it. I very much doubt that an option will be added
> 
> The biggest generation of errors I have is due to having Wi-Fi enabled and the Nintendo connections being blocked, which cannot connect in the background... Is there a way to avoid this?

using airplane mode, the only way I found

beyondmeat, 2025-02-15:
It would be nice for an option to filter out and don't log errors related to nintendo servers being blocked. If that can't be done, an option to disable the logs would be nice. Those that disable the logs wouldn't be filing issues and expecting help without logs, those would be typically advanced users who know better anyway.

closed 2025-04-03 by SciresM