Atmosphere-NX/Atmosphere #2440: 2162-0002 (0x4a2)

MintyHeal, 2024-11-07:
## Bug Report

### What's the issue you encountered?

After rolling back hos 19 to hos 18, I encountered errors, scored and rearranged everything. When I start atmosphere after a few seconds, I get this: [image], in full stock: [image]. 

### Crash Report
https://gist.github.com/MintyHeal/d632ae98af488cf2f1cd6331de22fa30#file-gistfile1-txt

### System Firmware Version

18.1.0</br>

### Environment?
- Hekate
- Unofficial build(https://github.com/Ultra-NX/UltraNX)
- EmuMMC and SysNAND:
  - EmuMMC file-based.

### Additional context?

As explained to me, this happened because I did not reset the old settings, I do not have a backup, the system may shut down after about 15 seconds, after which I receive error messages.

borntohonk, 2024-11-07:
> * Unofficial build(https://github.com/Ultra-NX/UltraNX)

Uh... your atmosphere crash report says -dirty and ontop of that you even list that...

Why did you even file this github issue here? https://github.com/Ultra-NX/UltraNX/issues

MintyHeal, 2024-11-07:
> Зачем вы вообще подали эту заявку на github здесь? https://github.com/Ultra-NX/UltraNX/issues
I applied here because I knew the problem and I needed help (the problem is not related to the build)

MintyHeal, 2024-11-07:
Solving the problem: holding down + - reset the system, and then go to full stock, format the cd, install the firmware

closed 2024-11-07 by MintyHeal