Atmosphere-NX/Atmosphere #2441: black screen after booting into atmosphere

LakeCharlesCountyDev, 2024-11-08:
accidentally updated to firmware 19.0 and my switch will not boot up into atmosphere. I've tried updating atmosphere, and hekate, and sigpatches, so far nothing works. if anyone comes across this post please help me out.

Whovian9369, 2024-11-08:
Piracy is not supported.

LakeCharlesCountyDev, 2024-11-08:
how is this related to piracy im just asking a question on how i can fix this issue.

LakeCharlesCountyDev, 2024-11-08:
i never mentioned anything about piracy

impeeza, 2024-11-08:
> i never mentioned anything about piracy

Because anything related to sigpatches is piracy. 

Also THIS IS NOT A SUPPORT site, if you need support go to Discord or GBATemp.

This site is only for report Atmosphère issues or bugs. This is not 

Please close and leave this post closed

LakeCharlesCountyDev, 2024-11-08:
well many other people have opened issues related to atmosphere and they have asked for support and recieved answers. IM ONLY STATING THAT I'VE UPDATED MY SIGPATCHES. I DO NOT NEED ANYTHING SIGPATCHES RELATED FROM ANYONE IM JUST ASKING FOR A SOLUTION.

closed 2024-11-08 by LakeCharlesCountyDev