Atmosphere-NX/Atmosphere #2444: Switch broken

TheButterGoblin, 2024-11-15:
This isn't necessarily a big report but my switch completely broke down after I tried downgrading the version so here is what happens I try to downgrade to the previous version and accidentally clicked only fat32 instead fat32 and exfat so now it will only boot into hekate and if I boot into anything else it shows the boot up screen and goes straight to black I don't want my switch to be broken if nobody has an answer I'm just going to bring it to my local rock 30 store and see if they can get it fixed.

impeeza, 2024-11-15:
Nothing to do with Atmosphère, if you need support please go to a support site like Discord or GBATemp.net

Please close the issue, this space IS ONLY TO REPORT BUGS AND PROBLEMS WITH ATMOSPHÈRE CODE.

By the way you should backup your card files AND THEN FORMAT THE CARD to FAT32 and restore files, that will solve your issue and prevent future corruptions.

TheButterGoblin, 2024-11-15:
Thanks for that

closed 2024-11-15 by TheButterGoblin