Atmosphere-NX/Atmosphere #2445: BSOD when logging into multiple player accounts while in a game

danistl, 2024-11-15:
## Bug Report

### What's the issue you encountered?

When playing any game where multiple people can log into their specific accounts, for example Mario Party Jamboree, it crashes in the account selection overlay when the second player clicks on their account picture.</br>

### How can the issue be reproduced?

Try to log into a second local account while playing a game.
Does not matter if the game was started with the first account or any other.</br>

### Crash Report

https://gist.github.com/danistl/d54afbc57b2989d6b38f299db1cd001e</br>

### System Firmware Version

19.0.1</br>

### Environment?

- hekate 6.2.2
- Atmosphère 1.8.0 pre-release
- SysNAND

**Installed Modules:**
Salt-NX
Tesla Overlay
FPSLocker
ReverseNX-RT
Status Monitor
emuiibo
EdiZon
sys-clk </br>

borntohonk, 2024-11-15:
isn't 0x37AC4C called initially from 0x44F3C which pertains to the escertificate/escommon/espersonalized/esmetarecord function
[image]

danistl, 2024-11-16:
> Does the issue occur if you remove all those modules, especially Tesla?

no difference without any modules

patriciocs1979, 2024-12-16:
I have the same issue

SciresM, 2025-04-03:
I never reproduced this, fwiw.

closed 2025-04-03 by SciresM