Atmosphere-NX/Atmosphere #2446: A Fatal Error Ocurred when running Fusee Unable to identify Package1!

jollivetamory, 2024-11-16:
I have an error message because I know the update of my switch to ver 19.0.1 and I have this message which appears how to resolve the problem

tommibru, 2024-11-17:
I had the same issue and the pre released version did allow me to lunch atmosphere. 

But my issue now is I cannot lunch anything once in atmosphere, none of the games nor tinfoil.

Any suggestions?

borntohonk, 2024-11-17:
> But my issue now is I cannot lunch anything once in atmosphere, none of the games nor tinfoil.

piracy is not supported.

kxtzownsu, 2025-01-04:
I'm getting this same error on latest Hekate, Atmosphere (pre-release) and sigpatches. My OFW is 19.0.1 as well, so I'm not sure why Atmosphere pre-release isn't working. I'm on Hekate 6.2.2 and Atmosphere 1.8.0-prerelease

Masamune3210, 2025-01-04:
Reading comprehension is not your strong suit, is it....

kxtzownsu, 2025-01-05:
> Reading comprehension is not your strong suit, is it....

How? I'm not trying to do any piracy, and I saw the one that said to use pre-release, which I did and it didn't work.

Masamune3210, 2025-01-05:
[image]
[image]

kxtzownsu, 2025-01-05:
> ![image](https://private-user-images.githubusercontent.com/1053504/400180343-af6e8c94-4092-4f78-b229-086985797df6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNTg5MjUsIm5iZiI6MTczNjA1ODYyNSwicGF0aCI6Ii8xMDUzNTA0LzQwMDE4MDM0My1hZjZlOGM5NC00MDkyLTRmNzgtYjIyOS0wODY5ODU3OTdkZjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDVUMDYzMDI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjExMGYyN2QyMzg3YzlkMWUxYTM3ZDllMmY3MzBiMzMwYmFmZTMxMzU0ZTA3NGQxYmUzOWY1ZjM1NDQ0ZGM3OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.CHG2WNnU8QK2kHjKfdpeaAX9Hb1-hSUdKmbAW32sxEI) ![image](https://private-user-images.githubusercontent.com/1053504/400180344-4bdd03e4-1294-4215-825a-6188f5326883.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNTg5MjUsIm5iZiI6MTczNjA1ODYyNSwicGF0aCI6Ii8xMDUzNTA0LzQwMDE4MDM0NC00YmRkMDNlNC0xMjk0LTQyMTUtODI1YS02MTg4ZjUzMjY4ODMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDVUMDYzMDI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWVlMjZjMWMxNjY0NjZjYTUzMzJmMmM0ZWZlY2M0MTcyZDlmNzllYWUwNGU0OWRhYjg0ODA1ZWNiOTA5YmM5OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.yUMcn4w0CosirOMcM3NSAXQS2M_HUUSygIMRbaFg_A0)

I'm gonna be :100: here even though it makes me seem like an idiot who just downloads whatever, I've never modded my Switch before this and I'm not *really* sure what sigpatches is. I know its my fault I didn't look it up, so my bad. If its piracy related I'll delete it. 🤷‍♂️

impeeza, 2025-01-05:
> > ![image](https://private-user-images.githubusercontent.com/1053504/400180343-af6e8c94-4092-4f78-b229-086985797df6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNTg5MjUsIm5iZiI6MTczNjA1ODYyNSwicGF0aCI6Ii8xMDUzNTA0LzQwMDE4MDM0My1hZjZlOGM5NC00MDkyLTRmNzgtYjIyOS0wODY5ODU3OTdkZjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDVUMDYzMDI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjExMGYyN2QyMzg3YzlkMWUxYTM3ZDllMmY3MzBiMzMwYmFmZTMxMzU0ZTA3NGQxYmUzOWY1ZjM1NDQ0ZGM3OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.CHG2WNnU8QK2kHjKfdpeaAX9Hb1-hSUdKmbAW32sxEI) ![image](https://private-user-images.githubusercontent.com/1053504/400180344-4bdd03e4-1294-4215-825a-6188f5326883.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwNTg5MjUsIm5iZiI6MTczNjA1ODYyNSwicGF0aCI6Ii8xMDUzNTA0LzQwMDE4MDM0NC00YmRkMDNlNC0xMjk0LTQyMTUtODI1YS02MTg4ZjUzMjY4ODMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDEwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMDVUMDYzMDI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZWVlMjZjMWMxNjY0NjZjYTUzMzJmMmM0ZWZlY2M0MTcyZDlmNzllYWUwNGU0OWRhYjg0ODA1ZWNiOTA5YmM5OCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.yUMcn4w0CosirOMcM3NSAXQS2M_HUUSygIMRbaFg_A0)
> 
> I'm gonna be 💯 here even though it makes me seem like an idiot who just downloads whatever, I've never modded my Switch before this and I'm not _really_ sure what sigpatches is. I know its my fault I didn't look it up, so my bad. If its piracy related I'll delete it. 🤷‍♂️

Yes those are piracy. Also this site is **not a support forum.** Please continue your questions on Discord or GBATemp

closed 2025-04-03 by SciresM