Atmosphere-NX/Atmosphere #2449: open retroarch_switch throw 0x4A8 (2168-0002) 010000000000100d

strivezsh, 2024-11-17:
## Bug Report

Atmosphère Crash Report (v1.7):
Result:                          0x4A8 (2168-0002)

Process Info:
    Process Name:                hbloader
    Program ID:                  010000000000100d
    Process ID:                  0000000000000085
    Process Flags:               000000b7
    User Exception Address:      00000025c970b000
Exception Info:
    Type:                        Data Abort
    Address:                     0000000000000000

### Crash Report
[01731820407_010000000000100d.log](https://github.com/user-attachments/files/17788973/01731820407_010000000000100d.log)

### System Firmware Version
Firmware 19.0.1
### Environment?
Hekate6.2.2
hbmenu3.6.0
EmuNand

tf fat32 1t

borntohonk, 2024-11-17:
this isn't really a bug, retroarch requires title-override (hold R on any game you have, which will launch hbmenu in high memory mode) then launch retroarch from there

strivezsh, 2024-11-17:
> this isn't really a bug, retroarch requires title-override (hold R on any game you have, which will launch hbmenu in high memory mode) then launch retroarch from there

you are right, it's work

closed 2024-11-17 by strivezsh

legionroy, 2024-11-22:
U save my life with this tip ty very much!