Atmosphere-NX/Atmosphere #2451: Error Code: 2168-0002 (0x4A8) 010000000000100d

QWERTYGVI, 2024-11-20:
Recently I downgraded my device's version to 16.1.0 Afterwards, I downgraded Atmosphere to 1.5.5. Other programs work, but when I run JKSV, I get an error: Error Code: 2168-0002 (0x4A8) Program: 010000000000100d The symptom persists even after downgrading the jksv version. I formatted the SD card and even changed the SD card, but the same symptom occurs. How can I fix this?

QWERTYGVI, 2024-11-20:
The reason I downgraded the version is to use LDN_mitm

impeeza, 2024-11-21:
Downgrade is never the solution, use the latest LDN_MiTM

https://github.com/DefenderOfHyrule/ldn_mitm/

impeeza, 2024-11-21:
also this is a repository of ATMOSPHÈRE not JSKV and your issue is FOR NONE OF THEM, please close the issue.

QWERTYGVI, 2024-11-21:
Sorry for the inconvenience. This is my first time using this feature on GitHub. Thanks for your reply.

closed 2024-11-21 by QWERTYGVI