Atmosphere-NX/Atmosphere #2452: Update ofw dualboot and then...

SereneMender133, 2024-11-20:
## Bug Report

### What's the issue you encountered?

Just updated ofw switch, when reboot it automatically reboot to the ofw, i tried holding - volume button STILL booted to ofw.

### How can the issue be reproduced?

Updating ofw

### Crash Report
N/A no crash only booted to ofw instead of atmosphere 

### System Firmware Version

19.0.1 updated

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Atmosphere don't launched.
- Homebrew software installed: no because ITS ofw but installed in Atmosphere
- EmuMMC or SysNAND:EmuMMC partition based
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

I used the HATS Pack.

Aura67, 2024-11-20:
Hats pack is a piracy package and Piracy is not Supported. Please close your Issues.

mr-f0xx, 2024-11-25:
@Aura67 what do you even mean by it's piracy package?

closed 2024-11-29 by SereneMender133