Atmosphere-NX/Atmosphere #2457: [Feature Request] Allow outputing higher resolutions / refresh rates

souldbminersmwc, 2024-11-23:
## Feature Request

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ If you do not have the information needed for a section, replace its contents with "Unknown". ]</br>
[ Lines between [ ] (square brackets) are to be removed before posting. ]

[ Please search for existing [feature requests](https://github.com/Atmosphere-NX/Atmosphere/issues?utf8=%E2%9C%93&q=is%3Aissue+label%3A%22features%2Ffeature-request%22) before you make your own request. ]</br>
[ Duplicate requests will be marked as such and you will be referred to the original request. ]

### What feature are you suggesting?
#### Overview:
Option which can be modified on-the-fly to enable higher resolutions/refresh rates

#### Smaller Details: 
L4T can do this. On OLED, 4k60 is possible, along with lower resolutions at higher refresh rates

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Addition
  - [ Ex: Addition of certain original features or features from other community projects. ]
  - [ If you are suggesting porting features or including features from other projects, include what license they are distributed under and what, if any libraries those project use. ]
- Change
- Removal
  - [Ex: Removal of certain features or implementation due to a specific issue/bug or because of low quality code, etc.]

### What component do you feel this would best fit within?
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
It would allow games to look better in 4k or higher refresh rates

masagrator, 2024-11-23:
This is completely out of scope for Atmosphere project. This can be done via homebrew, you don't need to incorporate it into Atmosphere

CasperMcFadden95, 2025-01-14:
https://github.com/Atmosphere-NX/Atmosphere/issues/2219

closed 2025-04-03 by SciresM