Atmosphere-NX/Atmosphere #2458: [Feature Request] Ram overclocking without kip patches

souldbminersmwc, 2024-11-23:
## Feature Request

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ If you do not have the information needed for a section, replace its contents with "Unknown". ]</br>
[ Lines between [ ] (square brackets) are to be removed before posting. ]

[ Please search for existing [feature requests](https://github.com/Atmosphere-NX/Atmosphere/issues?utf8=%E2%9C%93&q=is%3Aissue+label%3A%22features%2Ffeature-request%22) before you make your own request. ]</br>
[ Duplicate requests will be marked as such and you will be referred to the original request. ]

### What feature are you suggesting?
#### Overview:
Allow use of higher RAM OCs without kip patching. 1866 and 1996 are stable for all erista/mariko respectively, and with better ram types, you can go even further, for massive performance boost

#### Smaller Details: 
Some sort of config file?

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Addition
   Higher ram clocks without kip patches

### What component do you feel this would best fit within?
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
Allows higher ram clocks without needing to update loader.kip every AMS version.

masagrator, 2024-11-23:
Kip patching is necessary because otherwise there is no other way to do that than via reimplementing whole pcv sysmodule. And I can safely guess that reimplementing sysmodule responsible for hardware stability has very low priority.

impeeza, 2024-11-24:
SYS-Patch have the ability to patch any title at boot, Atmosphère do not condone piracy, you can go to GBATemp sys-patch thread to get support about this.  on the issue 18 of the hanai3Bi 's  Switch-OC-Suite github repo there is also a discussion about the topic.

souldbminersmwc, 2024-11-24:
???

closed 2025-04-03 by SciresM