Atmosphere-NX/Atmosphere #2459: Fatal error on boot.

Gamezpd, 2024-11-24:
## Bug Report

A fatal error occured when running atmosphere:
Program ID: 010041544d530000
Error desc: std::abort() called oxffe

Changed out fan inside switch. After rebuild, booted for a few seconds with switch logo, then crashed, now crashes on boot.

No changes

NA

Reproduced by booting switch with fusee. Other bin files like hekate dont init the sd card either, but the switch recognizes when one is plugged in. Full format of sd card has done nothing
[ * ]</br>

### Crash Report

Gist does not allow uploading of bin files, so heres a repository with the bin file:

https://github.com/Gamezpd/Fatal-switch-error

### System Firmware Version

19.0.0

### Environment?
Fusee, official, 18.0.0 atmos, no additionals, SysNand

### Additional context?

Formatted SD card to exFat and fat32, with fresh atmos install. Mo results, same crash on boot. Without SD card, requests SD card, with SD card, error code posted above is given, so SD card reader seems installed properly.

impeeza, 2024-11-24:
A quick search on Google: 

May be a sd corrupted or a hardware, anyway is not a Atmosphère issue, please search issues BEFORE create a new one:

https://gbatemp.net/threads/panic-occurred-while-running-atmosphere-010041544d530000.645978/

https://github.com/Atmosphere-NX/Atmosphere/issues/1378

https://github.com/Atmosphere-NX/Atmosphere/issues/813

Gamezpd, 2024-11-24:
> A quick search on Google:
> 
> May be a sd corrupted or a hardware, anyway is not a Atmosphère issue, please search issues BEFORE create a new one:
> 
> https://gbatemp.net/threads/panic-occurred-while-running-atmosphere-010041544d530000.645978/
> 
> #1378
> 
> #813

As I mentioned, the card has been reformated to fat32, fresh installs have been used of atmosphere, and the hardware detects the SD card correctly so both your suggestions of hardware and sd card issue seem invalid. I have seen both those threads which is why I explicitly mentioned what Ive done to troubleshoot those as not the issue.

closed 2024-11-24 by Gamezpd