Atmosphere-NX/Atmosphere #2460: Opening HBMenu with Album not working

Austingabe13, 2024-11-25:
## Bug Report

### What's the issue you encountered?

Opening HBMenu via the album is not working.

Opening HBMenu via the album shows nothing and prompts to press **A** for "OK." After this, I cannot exit, and the only fix is to hard reset and reboot the console.  

This issue occurred after exiting *Mario Kart 8 Deluxe* and then trying to open the album.

---

### How can the issue be reproduced?

1. Open a game.  
2. Return to the Home menu.  
3. Click on the album to launch HBMenu.  

---

### Crash Report

  
No crash report was generated.

---

### System Firmware Version

**Firmware Version:** 1.19.1  

---

### Environment?

- **Bootloader:** Hekate CTCaer 6.2.2  
- **Atmosphère Build:** Official release  
- **Additional kips or sysmodules loaded:**
  - MissionControl  
  - sys-clk (default clock)  
  - bootsound  
  - Everything in HATS Pack 1.8.0 Prerelease 1.2  

- **Homebrew software installed:** Yes  
- **Using EmuMMC or SysNAND:** SysNAND  

---

### Additional Context

- No custom themes installed.  
- Album launch works sometimes but causes this issue intermittently.

mr-f0xx, 2024-11-25:
Just make a nsp forwarder

borntohonk, 2024-11-25:
> Just make a nsp forwarder

piracy is not supported

Austingabe13, 2024-11-26:
Resolved via a reinstall of atmosphere

closed 2024-11-26 by Austingabe13