Atmosphere-NX/Atmosphere #2461: Error code 2002-4144 (0x206002)

ElochkaDev, 2024-11-27:
I do not understand at all what to do, it does not launch the atmosphere into any kind of both from the hackate and through the payload

[image]

borntohonk, 2024-11-27:
> I do not understand at all what to do, it does not launch the atmosphere into any kind of both from the hackate and through the payload

you deleted the template that you're supposed to fill out, go back and fill it out

ElochkaDev, 2024-11-27:
> > I do not understand at all what to do, it does not launch the atmosphere into any kind of both from the hackate and through the payload
> 
> you deleted the template that you're supposed to fill out, go back and fill it out edit: I assume you didn't fill it out because you are using an atmosphere fork // piracy package, it clearly says -dirty in your picture)

What is the template?

masagrator, 2024-11-27:
> What is the template?

The thing you deleted when writing your issue

SciresM, 2025-04-03:
fs::ResultResultRomNonRealDataVerificationFailed(), this is corruption, make a new emummc or follow an unbricking guide.

closed 2025-04-03 by SciresM