Atmosphere-NX/Atmosphere #2463: Crash 2162-0002 on wifi connect in downgraded emummc created from ofw 19.0.1

antonwe, 2024-11-28:
## Bug Report

### What's the issue you encountered?

When creating an emummc on a switch where ofw is 19.0.1 (installed through regular ninty system update) and then downgrading said emummc (I've tested both to 19.0.0 and 18.1.0), the system crashes as soon as it connects to wifi. The error is 2162-0002 (0x4A2) with program id 010000000000002f (npns). I've recreated the problem on two separate switches, one with a ninty account linked and one without. Same result.
If the emummc is upgraded back to 19.0.1 with daybreak, the problem goes away.
The problem does NOT occur with an emummc created from an older ofw version which has been upgraded to 19.0.1 with daybreak and then downgraded again (!)

### How can the issue be reproduced?

- Update ofw to 19.0.1 through ninty system update
- create emummc
- downgrade emummc to 18.1.0 or 19.0.0 with daybreak
- reboot into emummc and connect to wifi

### Crash Report

https://gist.github.com/antonwe/1d045bf8698f6418566212eaa1c001ca

### System Firmware Version

19.0.0

### Environment?

- Hekate 6.2.2 (official unmodified)
- Atmosphere 1.8.0 (official unmodified)
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: none
- EmuMMC or SysNAND: file-based emummc

### Additional context?

blank_prodinfo_emummc is enabled and atmosphere/hosts/default.txt blocks everything nintendo-related. I have checked with tcpdump on my nat gateway and apart from receiving an ip address through dhcp, no traffic is registered from/to the switch at all. The crash happens immediately after the ip adress has been ack'ed through dhcp.
The crash also happens with atmosphere 1.7.1 and emummc firmware 18.1.0.

masagrator, 2024-11-28:
prodinfo blanking is the reason why it's crashing. It works properly only below iirc 18.0.0 or 17.0.0

antonwe, 2024-11-28:
> prodinfo blanking is the reason why it's crashing. It works properly only below 17.0.0, because FW update introduced check that results in system crash if it's not met.
> 
> Here are patches that prevent crash https://github.com/fruityloops1/nim-prodinfo-blank-fix

this is not the nim crash.

masagrator, 2024-11-28:
> > prodinfo blanking is the reason why it's crashing. It works properly only below 17.0.0, because FW update introduced check that results in system crash if it's not met.
> > Here are patches that prevent crash https://github.com/fruityloops1/nim-prodinfo-blank-fix
> 
> this is not the nim crash.

Still npns uses ssl services, so it may crash for the same reason.

antonwe, 2024-11-28:
> Still npns uses ssl services, so it may crash for the same reason.

tried, still crashes.

antonwe, 2024-11-28:
maybe something for @ITotalJustice to look into

ITotalJustice, 2024-11-28:
Nothing to do with me

antonwe, 2024-11-28:
> Nothing to do with me

might be worth adding to sys-patch if it isn't fixed in atmosphere, is what I meant

ITotalJustice, 2024-11-28:
> > Nothing to do with me
> 
> might be worth adding to sys-patch if it isn't fixed in atmosphere, is what I meant

Again, that tool has nothing to do with me.

impeeza, 2024-11-29:
> > Nothing to do with me
> 
> might be worth adding to sys-patch if it isn't fixed in atmosphere, is what I meant

PIRACY IS NOT SUPPORTED, PLEASE CLOSE THE ISSUE.

antonwe, 2024-11-29:
> > > Nothing to do with me
> > 
> > 
> > might be worth adding to sys-patch if it isn't fixed in atmosphere, is what I meant
> 
> PIRACY IS NOT SUPPORTED, PLEASE CLOSE THE ISSUE.

this issue had absolutely nothing to do with piracy.

borntohonk, 2024-11-29:
> > > > Nothing to do with me
> > > 
> > > 
> > > might be worth adding to sys-patch if it isn't fixed in atmosphere, is what I meant
> > 
> > 
> > PIRACY IS NOT SUPPORTED, PLEASE CLOSE THE ISSUE.
> 
> this issue has absolutely nothing to do with piracy.

The homebrew you namedropped is, you made this issue become piracy related.

antonwe, 2024-11-29:
> > > > > Nothing to do with me
> > > > 
> > > > 
> > > > might be worth adding to sys-patch if it isn't fixed in atmosphere, is what I meant
> > > 
> > > 
> > > PIRACY IS NOT SUPPORTED, PLEASE CLOSE THE ISSUE.
> > 
> > 
> > this issue has absolutely nothing to do with piracy.
> 
> The homebrew you namedropped is, you made this issue become piracy related.

That homebrew solves another crash due to prodinfo blanking. Prodinfo blanking is an atmosphere feature. It makes sense to put it there if it wouldn't be fixed in atmosphere. Your argument is ridiculous. I am not pirating, I just want the crash fixed.

borntohonk, 2024-11-30:
> That homebrew solves another crash due to prodinfo blanking. Prodinfo blanking is an atmosphere feature. It makes sense to put it there if it wouldn't be fixed in atmosphere. Your argument is [...]

https://github.com/Atmosphere-NX/Atmosphere/issues/2215#issuecomment-1775462393

antonwe, 2024-11-30:
> [#2215 (comment)](https://github.com/Atmosphere-NX/Atmosphere/issues/2215#issuecomment-1775462393)

Interesting. The "nice thing" about that crash though is that it can easily be avoided by blocking ninty in the atmosphere hosts file. I hadn't yet found a way to prevent the crash in this issue, though, but thanks to you I just did.

> I checked your crash reports and found something referencing penne, which is a type of pasta, which i suppose is some sort of joke that it's "spaghetti code".
> 
> which apparently is related to system saves

This was an extremely helpful hint.

**My solution: enter airplane mode, start JKSV, find system save 00000110, title options -> open in file mode, delete penne_persistent.bin, close jksv, disable airplane mode and connect to wifi. Crash gone.**
THANK YOU

Deleting the entire save in JKSV doesn't seem to work, nothing happens when pressing A to delete it. But opening the save and deleting the penne file did the trick. @J-D-K

I would guess that this non-backwards-compatible save is created on ofw when the npns service manages to connect to ninty on 19.0.1, as I couldn't reproduce the crash by upgrading an older emummc to 19.0.1 - assumingly because ninty communication was blocked.

Spexs99, 2024-11-30:
> Interesting. The "nice thing" about that crash though is that it can easily be avoided by blocking ninty in the atmosphere hosts file. I hadn't yet found a way to prevent the crash in this issue, [...]

Hi there, I am encountering the same error as you are after following the same update and use of daybreak steps as you detailed in your first post. 

My device was on 19.0.1, daybreak to 18.1.0 and was getting the same atmosphere error. Reading this thread led me to use jksv but I am also having the issue in that I cannot delete the save penne_persistent.bin. 
When I click a to delete nothing happens. All other functions seem to be working fine.

I am being sure to launch in full memory mode with a game and I'm navigating to the correct part of the file system and using file view mode. I'm really not sure what step I am doing incorrectly. 

In the end after multiple installs and reinstalls I decided to used daybreak to do a reset to factory settings on the same version. 18.1.0 to 18.1.0 factory. I assumed that cleared the save file for me and was correct when using jksv to verify. 

just posting this as my alternate much less satisfying solution

antonwe, 2024-11-30:
@Spexs99 you may have to enable settings -> Enable Writing to System Saves" in jksv first.

borntohonk, 2024-11-30:
> Deleting the entire save in JKSV doesn't seem to work, nothing happens when pressing A to delete it. But opening the save and deleting the penne file did the trick. @J-D-K

if that is the case, then the return of 0xB9CEC call for return function at 0xDA090 is responsible

which has strings referencing:
"$.comb_db"
"$.accs"

which at 0xDA404 calls for function 0x5A90, and then that function at 0x5AAC calls for function 5A70 which has addres 0x5A84, which is in both threads of your crash report and in your fatal report.

ReturnAddress[02]:       0000001f3ce05a84 (npns + 0x5a84)

ann4belle, 2024-12-08:
> > > > > Nothing to do with me
> > > > 
> > > > 
> > > > might be worth adding to sys-patch if it isn't fixed in atmosphere, is what I meant
> > > 
> > > 
> > > PIRACY IS NOT SUPPORTED, PLEASE CLOSE THE ISSUE.
> > 
> > 
> > this issue has absolutely nothing to do with piracy.
> 
> The homebrew you namedropped is, you made this issue become piracy related.

By this logic, anyone can force an issue to be closed simply by mentioning *any* tool that enables piracy.

J-D-K, 2024-12-08:
> Deleting the entire save in JKSV doesn't seem to work, nothing happens when pressing A to delete it.
It's like that on purpose. I've learned over the past 8 years with JKSM and JKSV that people like to blame you for their own mistakes. Believe me. It gets old having people DM you to complain that they didn't know delete meant delete. I'm glad it got you out of a bind, but it's staying like that.

Leerz, 2024-12-09:
> Interesting. The "nice thing" about that crash though is that it can easily be avoided by blocking ninty in the atmosphere hosts file. I hadn't yet found a way to prevent the crash in this issue, [...]

verified 20241209 +8 15:29 - this solution works. 
also if you cannot delete, it means you haven't enabled system save delete.

jcxgarrido, 2025-01-10:
> @Spexs99 you may have to enable "settings -> Enable Writing to System Saves" in jksv first.
> 
> **Corrected solution: enter airplane mode, start JKSV, make sure "Enable Writing to System Saves" is enabled in settings, find system save 00000110, title options -> open in file mode, delete penne_persistent.bin, close jksv, disable airplane mode and connect to wifi.**

Thanks for everyone who contributed in resolving this. This was a lifesaver

idkcode22, 2025-02-11:
> [@Spexs99](https://github.com/Spexs99) you may have to enable "settings -> Enable Writing to System Saves" in jksv first.
>
> **Corrected solution: enter airplane mode, start JKSV, make sure "Enable [...]

TYSM!!!! I've been trying everything to fix this for 4hrs and this was the answer Thank YOU!!!!

closed 2025-04-03 by SciresM