Atmosphere-NX/Atmosphere #2464: Error Code 2002-3005 (0x177a02)

GSBro, 2024-12-03:
## Bug Report

### What's the issue you encountered?

Atmosphere 1.8.0 crashes on boot up 

### How can the issue be reproduced?

Launch Atmosphere 1.8.0 on Switch Firmware 19.0.0

### Crash Report

[image]

### System Firmware Version

19.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Atmosphere 1.8.0
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: ?
- EmuMMC or SysNAND: SysNAND I believe

### Additional context?

- Additional info about your environment:

When launching Atmosphere 1.8.0, it gets to the Switch logo screen, then it crashes.

borntohonk, 2024-12-03:
> ## Bug Report
> ### What's the issue you encountered?
> 
> Atmosphere 1.8.0 crashes on boot up
> ### How can the issue be reproduced?
> 
> Launch Atmosphere 1.8.0 on Switch Firmware 19.0.0
> ### Crash Report
> 
> ![20241203_091437](https://private-user-images.githubusercontent.com/71909568/392071263-be5d81b0-4dc7-4224-9841-b4ec1543f92e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMyNDc1NjUsIm5iZiI6MTczMzI0NzI2NSwicGF0aCI6Ii83MTkwOTU2OC8zOTIwNzEyNjMtYmU1ZDgxYjAtNGRjNy00MjI0LTk4NDEtYjRlYzE1NDNmOTJlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAzVDE3MzQyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdiNTZkMTRjZTkyNDY5ODdjMzhlZmI5NzU3NmVmNTQ4MDk2OGZiYTJlNmZmNDJlZGNhYzcyM2VjOTIzZTc0YWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Tcrc1WCHxLmUCgDkm3CzfWiuptyeb9-QljVziXIie-A)
> ### System Firmware Version
> 
> 19.0.0
> ### Environment?
> 
>     * What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
> 
>     * Official release or unofficial build: Atmosphere 1.8.0
> 
>     * Do you have additional kips or sysmodules you're loading: No
> 
>     * Homebrew software installed: ?
> 
>     * EmuMMC or SysNAND: SysNAND I believe
> 
> 
> ### Additional context?
> 
>     * Additional info about your environment:
> 
> 
> When launching Atmosphere 1.8.0, it gets to the Switch logo screen, then it crashes.

disable incognito/cal0 blanker (hekate)/exosphere blanker

GSBro, 2024-12-06:
> > ## Bug Report
> > ### What's the issue you encountered?
> > Atmosphere 1.8.0 crashes on boot up
> > ### How can the issue be reproduced?
> > Launch Atmosphere 1.8.0 on Switch Firmware 19.0.0
> > ### Crash Report
> > ![20241203_091437](https://private-user-images.githubusercontent.com/71909568/392071263-be5d81b0-4dc7-4224-9841-b4ec1543f92e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMyNDc1NjUsIm5iZiI6MTczMzI0NzI2NSwicGF0aCI6Ii83MTkwOTU2OC8zOTIwNzEyNjMtYmU1ZDgxYjAtNGRjNy00MjI0LTk4NDEtYjRlYzE1NDNmOTJlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAzVDE3MzQyNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdiNTZkMTRjZTkyNDY5ODdjMzhlZmI5NzU3NmVmNTQ4MDk2OGZiYTJlNmZmNDJlZGNhYzcyM2VjOTIzZTc0YWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Tcrc1WCHxLmUCgDkm3CzfWiuptyeb9-QljVziXIie-A)
> > ### System Firmware Version
> > 19.0.0
> > ### Environment?
> > ```
> > * What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
> > 
> > * Official release or unofficial build: Atmosphere 1.8.0
> > 
> > * Do you have additional kips or sysmodules you're loading: No
> > 
> > * Homebrew software installed: ?
> > 
> > * EmuMMC or SysNAND: SysNAND I believe
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ### Additional context?
> > ```
> > * Additional info about your environment:
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > When launching Atmosphere 1.8.0, it gets to the Switch logo screen, then it crashes.
> 
> disable incognito/cal0 blanker (hekate)/exosphere blanker
> 
> for hekate: cal0blank=0
> 
> for exosphere.ini: blank_prodinfo_sysmmc=0 blank_prodinfo_emummc=0

It didn't work, but thanks for the suggestion.

SciresM, 2025-04-03:
fs::ResultOutOfRange()...this is going to be corrupted nand, follow an unbricking guide.

closed 2025-04-03 by SciresM