Atmosphere-NX/Atmosphere #2465: Blank home screen after changing system language from eng to spa

Darkkef, 2024-12-04:
hi nintendo switch oled running latest atmosphere cfw fw 18.1, we were playing jamboree with the kids everythhing was running ok, and then i changed system language to spa because kiddo wanted to play triangle strategy and still learning more complex english, after i rebooted the switch to emummc it goes to homescreen but is all in blank, cannot see anything besides power menu when long pressing power button and home menu when long pressing said button (that's how i know cfw is booting apparently to home screen), that all started when i changed system settings in menu from english to spanish in cfw, so i guess that's the problem,  i tried modifying override_config.ini in sd atmosphere folder (didn't find any config.ini files to mod as suggested by ai and some forums) in hopes of changing system language without actually booting, im thinking in replacing atmosphere folder but dont wanna do it before ruling out other workarounds, since i set even my ownfoil and dont want anything else to break, is there a workaround to change the system language back to english in any other way??

not crash report, as technically no crash happened.

using hekate last version, NEXT AIO distribution of atmosphere.
homebrews, just tinfoil, linkhalo besides the usual stuff. emummc is partition based.

borntohonk, 2024-12-04:
> using hekate last version, NEXT AIO distribution of atmosphere. homebrews, just tinfoil, linkhalo besides the usual stuff. emummc is partition based.

Piracy is not supported.

Darkkef, 2024-12-04:
> > using hekate last version, NEXT AIO distribution of atmosphere. homebrews, just tinfoil, linkhalo besides the usual stuff. emummc is partition based.
> 
> Piracy is not supported.

Yes, im using my own legal dumped games, and free open source homebrew

Masamune3210, 2024-12-05:
doesn't matter. using homebrew specifically designed for piracy.

borntohonk, 2024-12-05:
> > > using hekate last version, NEXT AIO distribution of atmosphere. homebrews, just tinfoil, linkhalo besides the usual stuff. emummc is partition based.
> > 
> > 
> > Piracy is not supported.
> 
> Yes, im using my own legal dumped games, and free open source homebrew

Tinfoil is a closed source homebrew application, requiring piracy patches to be used, for piracy activities. 

no need to lie.

Darkkef, 2024-12-09:
> Tinfoil is a closed source homebrew application, requiring piracy patches to be used, for piracy activities.
>
> no need to lie.

Alas, I'm using my own dumped games lol

Martmists-GH, 2024-12-09:
Uninstall tinfoil and any sigpatches, and see if you can reproduce the issue. If not, we can't help.

impeeza, 2024-12-10:
Atmosphère DO NOT SUPPORT PIRACY, PLEASE DO NOT TALK ABOUT PIRACY NOR PIRACY SOFTWARE HERE, CLOSE THE ISSUE PLEASE.

closed 2024-12-12 by Darkkef