Atmosphere-NX/Atmosphere #2467: [Feature Request] donor nro openers

glitch128, 2024-12-11:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Using the override_config.ini, you can set a title to open a specific NRO in title override

#### Smaller Details: 
- For adding the setting to the config.ini, I was thinking it could be something like:
```
[NRO Forwarder]
forward_program_id_0=0000000000000000
forward_nro_location_0=path/to/app.nro
```
#### Nature of Request:  

- Addition
 
### What component do you feel this would best fit within?

Unknown

### Why would this feature be useful?

This could be useful for people who used to use forwarders, but as they're broken, this could replace them, as well as add a way to add shortcuts to homebrew apps on sysCFW in a non-bannable way.

souldbminersmwc, 2024-12-15:
amazing idea

closed 2025-04-03 by SciresM

glitch128, 2025-04-20:
Wait, so was this accepted or denied?

impeeza, 2025-04-20:
Wonder the same, we must see the changes on code.

SciresM, 2025-04-20:
Rejected.