Atmosphere-NX/Atmosphere #2471: Atmosphere crashes sometimes when taking out of dock or placing in dock

juniperdev, 2024-12-21:
## Bug Report

### What's the issue you encountered?

On several occasions, Atmosphere would crash when the switch is inserted into a physical dock. </br>
I'm not sure if this issue also occurs when the switch is removed from the dock, however I have not seen </br>
that happen yet. No changes to Atmosphere were made before these crashes; they happened during regular </br>
play sessions. </br>

### How can the issue be reproduced?

To reproduce the issue, try placing the switch into a physical dock either on the home screen or during gameplay. </br>
The occurrence of this issue is seemingly sporadic to me. It seems to be reproducible with any dock, since I have seen </br>
this same issue occur with 4 separate switch docks. </br>

### Crash Report

Five crash reports over the course of 4 months:
https://gist.github.com/juniperdev/ec6d0696584f55c23405c81c91642585 </br>

### System Firmware Version

1.7.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build:
  - Offical release build v6.2.1

- Do you have additional kips or sysmodules you're loading: Noexes, sys-patch, Fizeau, SaltyNX, MissionControl
- Homebrew software installed: Breeze, JKSV, RetroArch, All-in-One Switch Updater, Avatool, Browse-NX, DBI, Fizeau, ftpd pro 3.1.0, Goldleaf, HB App Store, MemTesterNX, mhz, NX-Shell, NXThemes Installer, Switchfin, Switchident, sys-clk manager, Themezer-NX, Tinfoil, TinyMemBench
- EmuMMC or SysNAND:
  - File-based emuMMC

### Additional context?

- Additional info about your environment:
Seems to mainly occur while on the home menu, after inserting switch into dock. Does not seem to be a dock issue; has this </br>
issue with multiple separate docks; seems to also happen after long uptimes.

impeeza, 2024-12-22:
Maybe is a outdated SYSModule, ugrade all of them, mainly MissionControl

juniperdev, 2024-12-22:
@impeeza thanks for the suggestion, will update MissionControl and other sysmodules. I'll drop another comment on whether the issue persists at an appropriate time.

juniperdev, 2024-12-22:
@impeeza found that Fizeau implemented a bug fix similar to the issue described in this release: https://github.com/averne/Fizeau/releases/tag/2.4.3

An issue on the Fizeau github repo describes this problem similarly: https://github.com/averne/Fizeau/issues/62

I believe I had a Fizeau version predating v2.4.3; I have promptly updated it a version that contains the bug fix.

I'm going to assume that this is the issue of Fizeau rather than Atmosphere, I will post another bug report if the issue reoccurs.

**TL;DR Update Fizeau to version > 2.4.2 **

closed 2024-12-22 by juniperdev