Atmosphere-NX/Atmosphere #2472: Add coredump as GDB monitor command

trevor403, 2024-12-22:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Support coredump via dmnt2 or creport
- Make it possible to trigger a coredump using gdb

#### Smaller Details: 
- Use the application or homebrew mappings to generate dump segments
- Write to SD card to speed up the process of taking memory snapshots
- Allow for targeting running modules or just the running application

#### Nature of Request:  
- Addition
  - **Add a new monitor command like `monitor coredump` which enumerates all virtual memory mappings used by the running application and dumps them to an ELF style coredump**

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
This would allow users to quickly search code and data using static analysis instead of waiting for lengthy memory dumps over the network with the existing gdb method.

This method can use the fast SD card to write the coredump and quickly snapshot multiple gigabytes of memory if need be.

trevor403, 2025-01-25:
I have not had time to work on a PoC for this unfortunately.

SciresM, 2025-04-03:
Not sure whether this should remain open, since this would eventually be a PR, not an issue, anyway.

hexkyz, 2026-02-05:
Closing down as this should be a PR in the future.

closed 2026-02-05 by hexkyz