Atmosphere-NX/Atmosphere #2473: Things in  atmosphere/contents completely deleted

zenngit1234, 2024-12-25:
## Bug Report

### What's the issue you encountered?

when i booted for CFW everything in atmosphere/contents are deleted.

### How can the issue be reproduced?

1,get any BotW mod (i get this for linkle mod and relics of the past.)
2,unzip it(if its BNP, i didnt use BCML so use 7z or zip mod.)
3, put folder named 01007EF00011E000 and 01007EF00011E001 in zip or 7z of mod to atmosphere/contents.
4,boot atmosphere on sysnand and fusee.

### Crash Report

N.A

### System Firmware Version

19.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:fusee
- Official release or unofficial build: 
  - [ Official release version 1.8.0 pre-release

- Do you have additional kips or sysmodules you're loading:N.A
- Homebrew software installed:atmosphere
- EmuMMC or SysNAND:Sysnand

### Additional context?
N.A

masagrator, 2024-12-25:
```
Do you have additional kips or sysmodules you're loading:N.A
```
You don't decide about that. It may be related. Because if it's really not, then only response you can get is that sdcard is corrupted or fake.

impeeza, 2024-12-25:
> ## Bug Report
> ### What's the issue you encountered?
> 
> when i booted for CFW everything in atmosphere/contents are deleted.
> ### How can the issue be reproduced?
> 
> 1,get any BotW mod (i get this for linkle mod and relics of the past.) 2,unzip it(if its BNP, i didnt use BCML so use 7z or zip mod.) 3, put folder named 01007EF00011E000 and 01007EF00011E001 in zip or 7z of mod to atmosphere/contents. 4,boot atmosphere on sysnand and fusee.
> ### Crash Report
> 
> N.A
> ### System Firmware Version
> 
> 19.0.1
> ### Environment?
> 
>     * What bootloader (fusèe, hekate, etc) was Atmosphère launched by:fusee
> 
>     * Official release or unofficial build:
>       
>       * [ Official release version 1.8.0 pre-release
> 
>     * Do you have additional kips or sysmodules you're loading:N.A
> 
>     * Homebrew software installed:atmosphere
> 
>     * EmuMMC or SysNAND:Sysnand
> 
> 
> ### Additional context?
> 
> N.A

A). Is not a Atmosphère issue.
B). Piracy is not supported

Please close the issue

borntohonk, 2024-12-25:
> A). Is not a Atmosphère issue. B). Piracy is not supported
>
> Please close the issue

Not sure where you're getting that from... Modding games is a legitimate and genuine usecase of atmosphere.

Masamune3210, 2024-12-27:
@impeeza nothing in the issue mentions piracy or a tool relating to it?

@zenngit1234 its less likely this is a bug in atmos and more likely your SD card is either failing or has filesystem damage. Pull it out and check it in a computer.

zenngit1234, 2024-12-27:
> ```
> N.A
> ```
> 
> You shouldn't ignore that. It may be related. Because if it's really not, then only response you can get is that sdcard is corrupted or fake (with fake being the most probable answer)
> 
> Overall, not an Atmosphere issue.

i dont have any kips or smth

zenngit1234, 2024-12-27:
> @impeeza nothing in the issue mentions piracy or a tool relating to it?
> 
> @zenngit1234 its less likely this is a bug in atmos and more likely your SD card is either failing or has filesystem damage. Pull it out and check it in a computer.

thank you. i try format it.

zenngit1234, 2024-12-27:
but maybe it not probe, so i dont close this yet.

SciresM, 2025-04-03:
Atmosphere has no file deletion code, this is almost definitely a problem with your SD card.

closed 2025-04-03 by SciresM