Atmosphere-NX/Atmosphere #2474: Error when opening DBI, JKSV and Yall are nuts

vocaz, 2024-12-29:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

### What's the issue you encountered?

Can't open any of these applications without a hard crash of the 010000000000001F program with code 0x4a2. Started happening after my update to 19.0.0. Haven't changed anything regarding atmosphere. 

### How can the issue be reproduced?

Unsure.

### Crash Report

https://gist.github.com/vocaz/6237c9bacc672717401ad790155bd756

### System Firmware Version

19.0.1</br>

### Environment?

Hekate
- Official release or unofficial build: 
  - Official v 1.8.0
- Do you have additional kips or sysmodules you're loading:
- Don't believe so
- Homebrew software installed: 
- Unsure, if you have a method for obtianing a list of everything installed more than happy to provide.
- EmuMMC or SysNAND:
  - Emummc
  - SD Raw partition

### Additional context?

Have tried deleting all of atmosphere and bootloader and reinstalling as well as clearing all themes.

borntohonk, 2024-12-29:
> ## Bug Report
> 
> [ If any section does not apply, replace its contents with "N/A". ] [ Lines between [ ] (square brackets) should be removed before posting. ]
> ### What's the issue you encountered?
> 
> Can't open any of these applications without a hard crash of the 010000000000001F program with code 0x4a2. Started happening after my update to 19.0.0. Haven't changed anything regarding atmosphere.
> ### How can the issue be reproduced?
> 
> Unsure.
> ### Crash Report
> 
> https://gist.github.com/vocaz/6237c9bacc672717401ad790155bd756
> ### System Firmware Version
> 
> 19.0.1
> ### Environment?
> 
> Hekate
> 
>     * Official release or unofficial build:
>       
>       * Official v 1.8.0
> 
>     * Do you have additional kips or sysmodules you're loading:
> 
>     * Don't believe so
> 
>     * Homebrew software installed:
> 
>     * Unsure, if you have a method for obtianing a list of everything installed more than happy to provide.
> 
>     * EmuMMC or SysNAND:
>       
>       * Emummc
>       * SD Raw partition
> 
> 
> ### Additional context?
> 
> Have tried deleting all of atmosphere and bootloader and reinstalling as well as clearing all themes.

> Error when opening DB
Piracy is not supported

vocaz, 2024-12-29:
Sorry, Was unaware DBI was a piracy software, just tried one extra software to see if i could help track down the program causing the bug

impeeza, 2024-12-29:
The one you exposed is not an Atmosphère issue, and this is not a support site, for support please go to Discord or GBATemp.

Please close the issue.

Always you upgrade your firmware you need to upgrade Atmosphère AND ALL YOUR HOMEBREW

closed 2024-12-30 by vocaz