Atmosphere-NX/Atmosphere #2475: Black screen after Launching Sfw or normal version

MojoDog29, 2025-01-02:
It all starts with me accidentally updating a game online on switch and it auto restarts the console. im on the menu and and i try to launch sfw or normal version and it goes black screen i try to restart the console inject the payload open the menu and its black screen again after launching either sfw or any other version. i try to inject the payload when the black screen happens praying it does something and gives me the hos panic occurred error code 06 colour ####. i have tried just updating or chaning the atmosphere version but nothing changes im still stuck with the blackscreen after launching sfw (maybe im not updating something the way i should idk or adding the right stuff). my sd card has no problems i have been using this thing for a while

impeeza, 2025-01-02:
Sorry, this is not a support site, you are not describing a Atmosphère Issue or bug, please close this issue.

For support please go to Discord or GBATemp.net

borntohonk, 2025-01-02:
If you're using hekate, you have to update that too.

closed 2025-04-03 by SciresM