Atmosphere-NX/Atmosphere #2476: lan-play and ldn_mitm issues with Atmosphere 1.8.0

H4X0R46, 2025-01-02:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
I'm not able to play on lan-play servers using ldn_mitm. I'm not sure if this is an atmosphere issue or not, but I've used it before, but now that I'm on fw 19.0.1, it doesn't work. I show up on the lan-play website and everything as an open lobby, but get an error when trying to connect to someone or when they try to connect to me (they're also on 19.0.1).
[ Did you make any changes related to Atmosphère itself? ]</br>
No, I'm running the latest 1.8.0 build and a fork of ldn_mitm made for newer fw.
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
Basically try to join a lan-play server with newest fw and newest atmosphere cfw. I'm using DefenderOfHyrule's ldn-mitm fork.

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

19.0.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- I'm usking Hekate to boot the fusee payload for atmosphere.
- Official release or unofficial build: 
- Official build 1.8.0
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- No, ldn_mitm is my only module I'm running.
- Homebrew software installed: [ * ]
- Nothing noteable, my setup is pretty bare bones right now.
- EmuMMC or SysNAND:
- EmuMMC File Based
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?
I'm just really wondering if this issue is due to Atmosphere or fw 19.0.1 because I know atmosphere for this fw is still in kind of earlier stages.

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

masagrator, 2025-01-02:
This is completely Atmosphere unrelated. Tool needs to be fixed, not Atmosphere.

borntohonk, 2025-01-03:
> because I know atmosphere for this fw is still in kind of earlier stages.

No?

> I'm using DefenderOfHyrule's ldn-mitm fork.

Then contact that person and have him open up his github issue tracker for his fork, and then submit the issue where it belongs (on his fork). That's how that works. You don't file unrelated issues in other projects issue trackers.

H4X0R46, 2025-01-03:
> No?
>
>
> Then contact that person and have him open up his github issue tracker for his fork, and then submit the issue where it belongs (on his fork). That's how that works. You don't file unrelated [...]

I was unsure if this was an Atmosphere bug not playing nice or a bug with ldn directly 🙄 Keyboard warriors everywhere.

masagrator, 2025-01-03:
If you have problem with some game or app on some OS, you are going first to that OS customer support or to that app support?

The same analogy

masagrator, 2025-02-19:
LDN was not reimplemented as part of Atmosphere project, so issue is completely unrelated to Atmosphere. My point stands

SciresM, 2025-04-03:
This is a lan-play/ldn_mitm issue.

closed 2025-04-03 by SciresM