Atmosphere-NX/Atmosphere #2478: A fatal error occurred when running Fusee. Abort called, Ir=0x40035835

Rudiy88, 2025-01-05:
A fatal error occurred when running Fusee. Abort called, Ir=0x40035835
[image]

всем привет. Нужна помощь. Ошибка возникла после запуска в режиме Full-stock OFW

Теперь официальная прошивка не загружается. Черный экран. Atmósfera не загружается выдает ошибку выше. Ничего не обновлял. Зашел в OFW - увидел сообщение о недоступности магазина(бан).  Перезагрузил консоль, и теперь вот эта ошибка. Официальная тоже не грузится. Черный экран.

Rudiy88, 2025-01-05:
Проблему решил!

Rudik66, 2025-01-07:
Как исправил ошибку?

Rudiy88, 2025-01-07:
> Как исправил ошибку?

У вас возникла такая же? И если да, то в результате чего? Я конечно вам расскажу как решил.

Rudik66, 2025-01-09:
Консоль использую не часто, она разрядилась, включаю и вот это

Rudiy88, 2025-01-11:
> Консоль использую не часто, она разрядилась, включаю и вот это

Да, могу помочь с решением в телеграм.

Rudik66, 2025-01-12:
@rudikrubiroid , вот сюда если не тяжело напиши решение

dmrepey, 2025-03-10:
Такая же проблема. Поможете?

Rudiy88, 2025-03-14:
> Такая же проблема. Можете?

Решили проблему?

closed 2025-04-03 by SciresM

swashwa77, 2025-06-02:
Как вы решили эту проблему?