Atmosphere-NX/Atmosphere #2480: whole system freeze and can't wake up from sleep using emunand

fffsfdsf, 2025-01-07:
i'm using the oldest switch verson which is pached of RCM bug ,
using fw 19.0.1, offical atomosphere 1.8.0 and offical hekate 6.2.2,
using plugins as listed below:

tesla(nx-ovlloader) V1.0.7
tesla-menu V1.2.3 
SaltyNX V1.2.0
FPSLocker V2.0.0
ReverseNX-RT V2.0.0
Status Monitor V1.0.4
emuiibo V1.0.0
sys-con V0.6.5
MissionControl V0.12.0
sys-clk V2.0.0

the problem started when i update the FW to 19.0.1 for playing splatoon 3 
it works amazing before the update (originaly fw 18.x) and have no problem
after update, all things works fine, and when i pressed the power button to sleep,it doesn't wake up and have no response to any key
i press the power button to force poweroff and turn it back on, it enters hekate as usual,and the emunand system works,but still, it will 100% freeze after sleep

512GB sd card, and using pi pico for modding

please tell me if any more information is needed

borntohonk, 2025-01-07:
Your whole bloated list of sysmodules, most which you don't need, are not all up to date.

example sys-clk v2.0.1 exists

you can probably fix your issue by deleting the /atmosphere/contents/ folder as that would purge all the sysmodules that you most likely don't need.

fffsfdsf, 2025-01-07:
> Your whole bloated list of sysmodules, most which you don't need, are not all up to date.
> 
> example sys-clk v2.0.1 exists
> 
> you can probably fix your issue by deleting the /atmosphere/contents/ folder as that would purge all the sysmodules that you most likely don't need.

ok I just checked every plugins' github humepage and find out that sys-clk have this issue in the update log
i updated everything and it works
thanks,and i'll close this

closed 2025-01-07 by fffsfdsf