Atmosphere-NX/Atmosphere #2481: Bug update 19.0.1

Choukettas, 2025-01-08:
Hello Atmosphere team,

First of all, thank you for all your hard work on Atmosphere and your support for the community. I'm reaching out to request support for Switch firmware version 19.0.1 in a future update.

Recently, my Switch was automatically updated to version 19.0.1 due to a forced update, and since then, I've encountered compatibility issues when trying to run Atmosphere CFW. I understand that the current version of Atmosphere supports up to firmware 19.0.0, but not for 19.0.1.

Could you please provide an update on when Atmosphere might support 19.0.1? Any information on a timeline or possible workarounds would be greatly appreciated.

Thank you for your assistance and for everything you do!

Aura67, 2025-01-08:
Atmosphere 1.8.0 supports firmware 19.0.1. What Nintendo brought was an update without a reboot and there should be no problems with the Atmosphere version.

(https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.8.0-prerelease/)

Masamune3210, 2025-01-08:
I'm pretty sure all the update did was reorganize and update ngwords, aka, the banned words list

impeeza, 2025-01-08:
> Hello Atmosphere team,
>
> First of all, thank you for all your hard work on Atmosphere and your support for the community. I'm reaching out to request support for Switch firmware version 19.0.1 in a [...]

Mate, thanks a lot for your kind words, Sadly this is not a Support site, this space is only to inform about Atmosphère bugs.

For support please go to Discord or GBATemp, on GBATem there is a tutorial about upgrade firmware and Atmosphère.

please close this issue.

> Atmosphere 1.8.0 supports firmware 19.0.1. What Nintendo brought was an update without a reboot and there should be no problems with the Atmosphere version.
> 
> https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.8.0-prerelease

Exactly, the Rebootless update only upgraded the list of bad words. no other change breaking any homebrew has been discovered so far.

closed 2025-01-09 by SciresM