Atmosphere-NX/Atmosphere #2482: gives this error. What to do?

Madara1-lang, 2025-01-14:
[image]

Madara1-lang, 2025-01-14:
what to do?

Madara1-lang, 2025-01-14:
> > что же делать?
> 
> Поиск по запросу 524850534f53454d в системе отслеживания ошибок показывает, что это аппаратная ошибка.
> 
> Загружается ли он без атмосферы? Если да, то, вероятно, неисправна SD-карта/emummc
> 
> если он не загружается без atmosphere, значит, консоль мертва.
> 
> правка: судя по вашей фотографии, это Switch Lite (Mariko), так что вам, возможно, придётся связаться с тем, кто модифицировал его для вас.

The stock version includes everything works fine, but Atmosfer gives this error. What should I do?
I bought a Bu, I don't know who did the chip.

borntohonk, 2025-01-14:
> The stock version includes everything works fine, but Atmosfer gives this error. What should I do?
> I bought a Bu, I don't know who did the chip.

Unfortunate for you. I guess you need to buy a new console.

Switch lite (mariko) requires modchips to run CFW, whoever installed it to your console have messed up installation and broke the memory/RAM // didn't clean up after and it corroded, so on and so forth.

Madara1-lang, 2025-01-14:
> > Стандартная версия включает в себя всё, что работает нормально, но Atmosfer выдаёт эту ошибку. Что мне делать? Я купил Bu, не знаю, кто делал чип.
> 
> К сожалению для вас. Полагаю, вам придётся купить новую приставку.
> 
> Switch lite (mariko) требует чипов для запуска CFW. Тот, кто установил его на вашу консоль, испортил установку и повредил память/оперативную память // не почистил после этого, и она заржавела, и так далее.

Everything works fine in stock, PC sees SD card. How to fix the error? RAM solder everything works fine.

borntohonk, 2025-01-14:
> Everything works fine in stock, PC sees SD card. How to fix the error? RAM solder everything works fine.

Hardware error. Fix is hardware related. Buy new hardware. You have modified hardware, take back to seller/person who modchipped it.

Madara1-lang, 2025-01-14:
> К сожалению для вас. Полагаю, вам придётся купить новую приставку.
>
> Switch lite (mariko) требует чипов для запуска CFW. Тот, кто установил его на вашу консоль, испортил установку и повредил [...]

In Stock everything works fine. But when switching to atmospheric, this error pops up, do you understand?

borntohonk, 2025-01-14:
> In Stock everything works fine. But when switching to atmospheric, this error pops up, do you understand?

It is you who choose to not understand despite having been told 3 times now. Accept reality, or get better translation software.

impeeza, 2025-01-15:
> In Stock everything works fine. But when switching to atmospheric, this error pops up, do you understand?

Your Modchip is bad/wrong installed, when you go to stock the modchip have nothing to do, when you try to run CFW modchip needs to be full working.

Take your console to the modchip installer or change your modchip.

please close the issue because have nothing to do with Atmosphère.

Madara1-lang, 2025-01-15:
> > > > Стандартная версия включает в себя всё, что работает нормально, но Atmosfer выдаёт эту ошибку. Что мне делать? Я купил Bu, не знаю, кто делал чип.
> > > 
> > > 
> > > К сожалению для вас. Полагаю, вам придётся купить новую приставку.
> > > Switch lite (mariko) требует чипов для запуска CFW. Тот, кто установил его на вашу консоль, испортил установку и повредил память/оперативную память // не почистил после этого, и она заржавела, и так далее.
> > 
> > 
> > В Stock всё работает нормально. Но при переключении на атмосферную версию появляется эта ошибка, понимаете?
> 
> Это вы решили не понимать, несмотря на то, что вам уже трижды объяснили. Примите реальность или найдите более качественное программное обеспечение для перевода.

Я же говорил что с консолью все нормально.Если вы не разобравшись сделали вывод не надо его писать.А сначала уточнить в чём причина.Перезалил прошивку последней версией,отформативовав в Fat32 накатил новую и всё вауля.Работает.А не то что вы писали.Кирпич,покупать новую или дело с чипом.Спасибо за помощь которую вы не оказали .

Madara1-lang, 2025-01-15:
All I need to do is reinstall Atmosphere.

closed 2025-01-15 by Madara1-lang