Atmosphere-NX/Atmosphere #2483: Error Code: 2162-0002 (0x4a2) / 0100000....02F

MFr4nky, 2025-01-19:
## Bug Report

### What's the issue you encountered?

I created a modded switch(V2) and create the SDCard with all the files (Atmosphere,hekate,....), they are all the versions for the
FW19.0.0.

I had before another issue(0xffe) but i fixed it, due to overite the atmosphere and bootloader files.

### How can the issue be reproduced?

When i try to boot in the CFW(emummc)

### Crash Report

Atmosphère Fatal Report (v1.1):
Result:                          0x4A2 (2162-0002)

Program ID:                      010000000000002f
Process Name:                    npns
Firmware:                        19.0.0 (Atmosphère 1.8.0-master-c6014b533)
General Purpose Registers:
         FP:                     0000003ecdc67d20
         LR:                     000000022d433894
         SP:                     0000003ecdc67d20
         PC:                     000000022d446d70
Start Address:                   000000022d400000
Stack Trace:
        ReturnAddress[00]:       000000022d448aa4
        ReturnAddress[01]:       000000022d448364
        ReturnAddress[02]:       000000022d448498
        ReturnAddress[03]:       000000022d405bec
        ReturnAddress[04]:       000000022d405990
        ReturnAddress[05]:       000000022d405a64
        ReturnAddress[06]:       000000022d405a84
        ReturnAddress[07]:       000000022d405ab0
        ReturnAddress[08]:       000000022d4da408
        ReturnAddress[09]:       000000022d4b9cf0
        ReturnAddress[10]:       000000022d4d42f4
        ReturnAddress[11]:       000000022d4d421c
        ReturnAddress[12]:       000000022d4bf2a4
        ReturnAddress[13]:       000000022d4bbf3c
        ReturnAddress[14]:       000000022d4ba50c
        ReturnAddress[15]:       000000022d42b0d0
        ReturnAddress[16]:       000000022d42dda4
        ReturnAddress[17]:       0000000000000000
Stack Dump:                               00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             003ecdc67d20 80 7d c6 cd 3e 00 00 00 a4 8a 44 2d 02 00 00 00
                             003ecdc67d30 01 01 00 00 00 00 10 00 0c 00 00 80 01 00 00 00
                             003ecdc67d40 03 00 00 00 05 00 00 08 00 00 14 00 00 00 00 00
                             003ecdc67d50 68 7d c6 cd 3e 00 00 00 00 00 00 00 15 00 00 00
                             003ecdc67d60 80 06 63 2d 02 00 00 00 50 02 00 00 00 00 00 00
                             003ecdc67d70 30 7d c6 cd 3e 00 00 00 00 46 8d 9c 26 00 00 00
                             003ecdc67d80 f0 7d c6 cd 3e 00 00 00 64 83 44 2d 02 00 00 00
                             003ecdc67d90 48 58 50 2d 02 00 00 00 e8 7f c6 cd 3e 00 00 00
                             003ecdc67da0 48 58 50 2d 02 00 00 00 00 00 00 00 00 00 00 00
                             003ecdc67db0 08 04 63 2d 02 00 00 00 d0 04 63 2d 02 00 00 00
                             003ecdc67dc0 2c 7e c6 cd 3e 00 00 00 00 00 00 00 00 00 00 00
                             003ecdc67dd0 20 04 63 2d 02 00 00 00 10 1c 51 2d 02 00 00 00
                             003ecdc67de0 c0 04 63 2d 02 00 00 00 d0 04 63 2d 02 00 00 00
                             003ecdc67df0 30 7e c6 cd 3e 00 00 00 98 84 44 2d 02 00 00 00
                             003ecdc67e00 48 58 50 2d 02 00 00 00 c0 03 63 2d 02 00 00 00
                             003ecdc67e10 80 bd 62 2d 02 00 00 00 70 7e c6 cd 3e 00 00 00
TLS Address:                 000000269c8d4600
TLS Dump:                                 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             00269c8d4600 06 00 10 00 0c 00 00 80 01 00 00 00 1d 00 34 00
                             00269c8d4610 00 00 00 00 50 02 00 00 80 06 63 2d 00 00 00 20
                             00269c8d4620 53 46 43 49 01 00 00 00 02 00 00 00 00 00 00 00
                             00269c8d4630 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d4640 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d4650 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d4660 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d4670 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d4680 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d4690 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d46a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d46b0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d46c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d46d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d46e0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             00269c8d46f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

### System Firmware Version

19.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:

- hekate 6.2.2. and the same fusee.bin
- 
- Official release: 
 Atmosphere 1.8.0 (Github)

[image]

[image]

MFr4nky, 2025-01-19:
I tried the Stehtix Hat-Pack but the have the same issue.... can it be possible i f*** up the soldering ?

[image]

borntohonk, 2025-01-19:
> I tried the Stehtix Hat-Pack but the have the same issue.... can it be possible i f*** up the soldering ?
> 
> [image]

"hat pack", in other words, piracy package.

piracy is not supported.

support is also not given for modchips.

borntohonk, 2025-01-19:
Also you have the exact same issue as https://github.com/Atmosphere-NX/Atmosphere/issues/2463

solution is within the other github issue, and also upgrade to 19.0.1.

impeeza, 2025-01-19:
> ## Bug Report
> ### What's the issue you encountered?
> 
> I created a modded switch(V2) and create the SDCard with all the files (Atmosphere,hekate,....), they are all the versions for the FW19.0.0.
> 
> I had before another issue(0xffe) but i fixed it, due to overite the atmosphere and bootloader files.
> ### How can the issue be reproduced?
> 
> When i try to boot in the CFW(emummc) after few seconds in the Home-menu, same with the Stock(FW). It seems that it boot correctly in the CFW or FW but then it crashes. I also can not boot the original FW without SD.
> ### Crash Report
> 
> Atmosphère Fatal Report (v1.1): Result: 0x4A2 (2162-0002)
> 
> Program ID: 010000000000002f Process Name: npns Firmware: 19.0.0 (Atmosphère 1.8.0-master-c6014b533) General Purpose Registers: FP: 0000003ecdc67d20 LR: 000000022d433894 SP: 0000003ecdc67d20 PC: 000000022d446d70 Start Address: 000000022d400000 Stack Trace: ReturnAddress[00]: 000000022d448aa4 ReturnAddress[01]: 000000022d448364 ReturnAddress[02]: 000000022d448498 ReturnAddress[03]: 000000022d405bec ReturnAddress[04]: 000000022d405990 ReturnAddress[05]: 000000022d405a64 ReturnAddress[06]: 000000022d405a84 ReturnAddress[07]: 000000022d405ab0 ReturnAddress[08]: 000000022d4da408 ReturnAddress[09]: 000000022d4b9cf0 ReturnAddress[10]: 000000022d4d42f4 ReturnAddress[11]: 000000022d4d421c ReturnAddress[12]: 000000022d4bf2a4 ReturnAddress[13]: 000000022d4bbf3c ReturnAddress[14]: 000000022d4ba50c ReturnAddress[15]: 000000022d42b0d0 ReturnAddress[16]: 000000022d42dda4 ReturnAddress[17]: 0000000000000000 Stack Dump: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 003ecdc67d20 80 7d c6 cd 3e 00 00 00 a4 8a 44 2d 02 00 00 00 003ecdc67d30 01 01 00 00 00 00 10 00 0c 00 00 80 01 00 00 00 003ecdc67d40 03 00 00 00 05 00 00 08 00 00 14 00 00 00 00 00 003ecdc67d50 68 7d c6 cd 3e 00 00 00 00 00 00 00 15 00 00 00 003ecdc67d60 80 06 63 2d 02 00 00 00 50 02 00 00 00 00 00 00 003ecdc67d70 30 7d c6 cd 3e 00 00 00 00 46 8d 9c 26 00 00 00 003ecdc67d80 f0 7d c6 cd 3e 00 00 00 64 83 44 2d 02 00 00 00 003ecdc67d90 48 58 50 2d 02 00 00 00 e8 7f c6 cd 3e 00 00 00 003ecdc67da0 48 58 50 2d 02 00 00 00 00 00 00 00 00 00 00 00 003ecdc67db0 08 04 63 2d 02 00 00 00 d0 04 63 2d 02 00 00 00 003ecdc67dc0 2c 7e c6 cd 3e 00 00 00 00 00 00 00 00 00 00 00 003ecdc67dd0 20 04 63 2d 02 00 00 00 10 1c 51 2d 02 00 00 00 003ecdc67de0 c0 04 63 2d 02 00 00 00 d0 04 63 2d 02 00 00 00 003ecdc67df0 30 7e c6 cd 3e 00 00 00 98 84 44 2d 02 00 00 00 003ecdc67e00 48 58 50 2d 02 00 00 00 c0 03 63 2d 02 00 00 00 003ecdc67e10 80 bd 62 2d 02 00 00 00 70 7e c6 cd 3e 00 00 00 TLS Address: 000000269c8d4600 TLS Dump: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f 00269c8d4600 06 00 10 00 0c 00 00 80 01 00 00 00 1d 00 34 00 00269c8d4610 00 00 00 00 50 02 00 00 80 06 63 2d 00 00 00 20 00269c8d4620 53 46 43 49 01 00 00 00 02 00 00 00 00 00 00 00 00269c8d4630 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d4640 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d4650 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d4660 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d4670 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d4680 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d4690 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d46a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d46b0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d46c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d46d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d46e0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00269c8d46f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> ### System Firmware Version
> 
> 19.0.0
> ### Environment?
> 
>     * What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
> 
>     * hekate 6.2.2. and the same fusee.bin
> 
>     * Official release:
>       Atmosphere 1.8.0 (Github)
> 
> 
> [image]
> 
> [image]

First: THIS IS NOT A SUPPORT SITE, please go to Discord for support.
SECOND: do not open a issue for a well know problem caused by configuration error,  READ THE DOCUMENTATION AND CONFIG YOUR SETUP PROPERLY.
THIRD: piracy is not supported 

Please close this issue, as is not a Atmosphère issue

MFr4nky, 2025-01-19:
Ok, sorry and thx!

closed 2025-01-19 by MFr4nky