Atmosphere-NX/Atmosphere #2486: Please update atmosphere to 19.0.1 firmware

SwitchAdventurer, 2025-01-23:
Please update this to 19.9.1 :C I suddently updated the switch for error and now I don't have atmosphere, what I can do?

Aura67, 2025-01-23:
Atmosphere supports firmware 19.0.1, no update needs to be written. Nintendo only did a rebootles update where words were updated without a reboot.

https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.8.0-prerelease

SwitchAdventurer, 2025-01-23:
But so why I don't have it anymore? Like I don't have anymore the hb channel

shikariunite, 2025-01-23:
Nothing. 19.1.1 isn't yet supported and has been in beta for a while, finding bugs before it's properly released.

masagrator, 2025-01-23:
> Nothing. 19.1.1 isn't yet supported and has been in beta for a while, finding bugs before it's properly released.
> […](#)

Please don't spread misinformation, thank you. Newest FW is 19.0.1, is properly supported and latest release is not beta.

SwitchAdventurer, 2025-01-23:
Also I wronged I wanted tò meaning the 19.0.1 firmware

InfinityCoda, 2025-01-23:
If you are receiving an error within Atmosphere, you will need to post the details of that error to get any help with it. Vague statements like "why it doesn't work?" give no useful info. But based on:

> now I don't have atmosphere
> But so why I don't have it anymore? Like I don't have anymore the hb channel

Sounds like you simply haven't booted into atmosphere. Check System Settings where your software version is listed, an AMS version should be present as well. If not then you will need to inject a payload through your preferred method and boot Atmosphere. This is not an Atmosphere bug and support should be sought elsewhere such as the Discord.

SwitchAdventurer, 2025-01-24:
No, now I resolved, the errore was that I didn't install the sigpatches

Masamune3210, 2025-01-24:
..................

Aura67, 2025-01-24:
> No, now I resolved, the errore was that I didn't install the sigpatches

Piracy is not supported please delete atmosphere from your sd card and download it again and unpack it to your sd card.

SwitchAdventurer, 2025-01-24:
Wait, are the sigpatch piracy? I wasn't about this thing

closed 2025-01-24 by SwitchAdventurer