Atmosphere-NX/Atmosphere #2488: Atmosphere fatal error when launching title. EmuMMC not launchable ever since

Semsem04, 2025-02-03:
## Bug Report

### What's the issue you encountered?

 Launched XC3 and got a fatal error occurred and a bin file was saved. No changes were made to atmosphere directly but that day I installed OC-Switchcraft-EOS and overclocked the system including changing RAM timings, CPU/GPU Undervolting. EmuMMC can't be launched bc screen goes black after the atmosphere logo. OFW is fine

### How can the issue be reproduced?

Not entirely sure what caused this

### Crash Report

Uploaded to mediafire: https://www.mediafire.com/file/2tetca9elp2b8ng/report_000000155da682ec.bin/file

### System Firmware Version

19.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Official 1.8.0
- Do you have additional kips or sysmodules you're loading: OC-Toolkit, sys-clk,sys-tweak,sys-patch 
- Homebrew software installed: Only most basic stuff
- EmuMMC or SysNAND: Partition based EmuMMC

### Additional context?

No

borntohonk, 2025-02-03:
> * sys-patch

Piracy is not supported.

also the bundle you use is not atmosphere from this repository, it's altered and loads in external loader.kip among other things. You should contact whoever runs "OC-Switchcraft-EOS"

Semsem04, 2025-02-03:
Thanks for the answer.
I already contacted them and I did the usual fixes but that didnt work. Even without loading the loader.kip it doesn't work.
I'd only like for somebody to tell me what the error file states so I can fix it.
If that doesn't work then its most likely EmuMMC corruption but I'd like to at least try

hakuboo123, 2025-02-05:
Hello I’m having the same issue. First it was atmosphere error now it’s kips. Was working fine then out of no where EmuMMC red screened then all the following.

borntohonk, 2025-02-05:
> now it’s kips.

Those are not wildcard loaded by atmosphere/fusee, That's a hekate feature.

File an issue with the hekate repository, or seek support from whoever is supplying you with the .kips which you are loading in for reasons unknown.

This github issue doesn't belong here.

Semsem04, 2025-02-05:
Initial issue can be written off as EmuMMC corruption due to overclocking and undervolting etc. 
Deleted and created a new EmuMMC and installed the same softwares with safer clock-settings and everything is still working.
Would've been grateful if at least somebody could've told me what the error report states but that's irrelevant now.
Closing this issue

closed 2025-02-05 by Semsem04