Atmosphere-NX/Atmosphere #2489: A fatal error occured when running Atmosphère. Program ID: 524850534f53454d

CodigoHacendado, 2025-02-06:
## Bug Report

### What's the issue you encountered?

The system crash while dockered and in standby unplug and plug the power adapter.

### How can the issue be reproduced?

I really don't know.

### Crash Report

[report_0000009cb7e19684.zip](https://github.com/user-attachments/files/18695162/report_0000009cb7e19684.zip)

### System Firmware Version

19.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee.bin
- Official release or unofficial build: Official release version 1.8.0
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: EmuMMC partition-based

### Additional context?

N/A

borntohonk, 2025-02-07:
> ## Bug Report
> ### What's the issue you encountered?
>
> The system crash while dockered and in standby unplug and plug the power adapter.
> ### How can the issue be reproduced?
>
> I really don't know.
> ### [...]

Quick search for "524850534f53454d" -> Hardware error // corrupt emummc.

try new sd card, new emummc, new setup

if still problem, new console.

SciresM, 2025-02-07:
This seems like a one-off hardware error.

closed 2025-02-07 by SciresM