Atmosphere-NX/Atmosphere #2490: PANIC OCCURED WHILE RUNNING ATMOSPHERE.

Pwndflip, 2025-02-07:
## Bug Report

### What's the issue you encountered?

When i booted into hekate, on my modchipped nintendo switch (Picofly chip) , i went into atmosphere, my custom firmware installed. At first, it booted without a problem, but now, when i try to open it, it gives me the error, PANIC OCCURED WHILE RUNNING ATMOSPHERE.

### How can the issue be reproduced?

[boot into hekate, press on CFW on MicrosdCard, see the Atmosphere logo, get an Error.

### Crash Report

[report_05acc477.zip](https://github.com/user-attachments/files/18709692/report_05acc477.zip)

### System Firmware Version

19.0.1</br>

### Environment?

- Bootloader: Atmosphère launched by Hekate.
- Official release or unofficial build: 1.8.0
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: Yes
- EmuMMC or SysNAND: EMUMMC / Partition Based.
  

### Additional context?

- Additional info about your environment: No.

SciresM, 2025-02-07:
Fix your archive bits using the tool in hekate.

closed 2025-02-07 by SciresM

Pwndflip, 2025-02-07:
> Fix your archive bits using the tool in hekate.
 
Okay, didnt know that is was that easy😅

Thank you very much for your help Bro!
Appreciate it!