Atmosphere-NX/Atmosphere #2491: A fatal error occurred when running Atmosphere Program ID:0100000000000005

ButtBoyBoris, 2025-02-09:
## Bug Report

### What's the issue you encountered?

I have a unpatched Switch that got updated and after this I get the fatal error "A fatal error occurred when running Atmosphere Program ID:0100000000000005" and it reboots to a blackscreen. I have tried downloading a clean version of Atmosphere and the latest version of Hekate, but still the same result.

### How can the issue be reproduced?

Boot CFW gives the same result over and over again.

### Crash Report

https://gist.github.com/ButtBoyBoris/fa6fc60f5bec6c6f9d34a392dae81a5c

### System Firmware Version

19.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release:
  - Using hekate v6.2.2 & Nyx v1.6.4 to boot Atmosphere, have also tried injection Fusee with the same result.
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: None
- EmuMMC or SysNAND: SysNAND (Same result with EmuMMC)
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- When I try to boot OFW I only get a black screen and I have checked the motherboard for any hardware faults, cant find any.

borntohonk, 2025-02-10:
> * When I try to boot OFW I only get a black screen and I have checked the motherboard for any hardware faults, cant find any

Use hekate; Go to "Console Info" Tab, then eMMC, then run a benchmark, and/or verify the partition table is intact.

If it isn't intact, now would be the time you would normally bring out the backup of your emmc you should have. (and potentially considering ordering a spare part to flash your backup to)

borntohonk, 2025-02-11:
> I have tried to rebuild the emmc on a donor NAND

No actual backup of your boot0/boot1/emmc?

https://suchmememanyskill.github.io/guides/unbrick/

SciresM, 2025-04-03:
i2c::ResultNoAck() -- this is a hardware issue.

closed 2025-04-03 by SciresM