Atmosphere-NX/Atmosphere #2492: Redirecting switch internal server address

Unagaml, 2025-02-26:
## Feature Request

### What feature are you suggesting?
Redirecting switch internal server address
#### Overview:
- I want to see a function like pretendo that allows you to redirect requests to your servers. For example, I am writing a large project on servers and I really need a patch so that, for example, a request that goes to dauth-%.ndas.srv.nintendo.net goes to the address dauth-lp1-ndas-srv-hekentdo.ru</br>

#### Smaller Details: 
- and also so that it would be possible to turn on or off the UltaHand menu</br>

### Why would this feature be useful?]
This feature will finally allow people with modified consoles to play online without crutches and create their own servers</br>
Creation of your own servers to simulate full online and even greater customization of the system)</br>

borntohonk, 2025-02-26:
> * For example, I am writing a large project on servers and I really need a patch so that, for example, a request that goes to dauth-%.ndas.srv.nintendo.net goes to the address dauth-lp1-ndas-srv-hekentdo.ru

This is already a thing, and the feature is implemented as "dns.mitm"
https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/features/dns_mitm.md

> * and also so that it would be possible to turn on or off the UltaHand menu

Uh... That's not an atmosphere feature in the first place, if you thought it was, then this probably is the wrong repository to submit this github issue.

> ### Why would this feature be useful?
> 
> This feature will finally allow people with modified consoles to play online without crutches and create their own servers Creation of your own servers to simulate full online and even greater customization of the system)

People with atmosphere can go online at anytime, even eshop, risk free, completely safe, no crutches. You just have to pay for NintendoSwitchOnline (NSO) subscription.

The only use-case where this isn't true, are for people who break the law and pirate things or install things they really shouldn't (impossible with just atmosphere anyway).

Unagaml, 2025-02-26:
> This is already a thing, and the feature is implemented as "dns.mitm"
> https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/features/dns_mitm.md
>
>
> Uh... That's not an atmosphere feature in [...]

Okay, I'll check the performance of this function, if anything, I'll write

closed 2025-04-03 by SciresM